libopencm3/libopencm3 #1276: gpio_set_mode() for F1 and gpio_mode_setup() for anyone else in STM32

tute-avalos, 2020-12-14:
Why are we using diferent API funtions for F1 in STM32? I suggest to choose one (`gpio_mode_setup()`) as the standard one, adding a macro for `gpio_set_mode()` as legacy one with _"Deprecated WARNING"_ at compile time. And, in sometime, call it a year or two, finally erase It.

manuelbl, 2020-12-14:
Quote from [#266](https://github.com/libopencm3/libopencm3/issues/266#issuecomment-34876423):

> The gpio modes is an altogether different story though. The gpio peripheral (and the related alternate function settings) are completely different on the f1. (And no-body likes them)
> 
> It's _possibly_ worth some very simple lowest common denominator function ala Arduino's "pinMode(GPIOB, 11, OUTPUT)" but you're always going to be missing things there, or at least, it's not as easy a change.

karlp, 2020-12-15:
the nottes in #266 are right, but it is still feasible to make an improvement.  I doubt it will end up with deprecations or anything though.  This is a known issue, so I'm not going to keep a tracking ticket around for it, but pull requests attempting to provide a common ground api would be reviewed.  (but do note that it is not just as simple as changing the name, they _are_ different beasts)

closed 2020-12-15 by karlp