libopencm3/libopencm3 #1278: STM32F4xx External interrupt on pins other than 0-4

katy-kem, 2020-12-15:
Hi. I have a board with an STM32F4xx and I'd like to use GPIOC8 as an external interrupt. I cannot get the code to work. I can get it to work if I use GPIOC3, but this is not ideal with the existing board. Is it possible to use a pin 8 as an external interrupt on an STM32F4 with libopencm3? 

According to [this site](https://stm32f4-discovery.net/2014/08/stm32f4-external-interrupts-tutorial/) the STM32F4 is capable of external interrupts on a pin 8. The interrupt handler for this using the STM32CubeIDE is called `EXTI9_5_IRQn` as it runs interrupts on pins 5-9. According to the [EXTI declares section of the libopencm3 wiki](http://libopencm3.org/docs/latest/stm32f4/html/group__exti__defines.html), there are many EXTIs that can be declared, including an EXTI8. However, according to the [NVIC section of the wiki](http://libopencm3.org/docs/latest/stm32f4/html/stm32_2f4_2nvic_8h.html), there are only NVIC_EXTIx_IRQ 's for x=0-4. 

When I set up the code for GPIOC3 as the interrupt pin:

```
static void exti_setup(void)
{

	/* Enable EXTI3 interrupt. */
	nvic_enable_irq(NVIC_EXTI3_IRQ);

	/* Set GPIO0 (in GPIO port A) to 'input open-drain'. */
	gpio_mode_setup(GPIOC, GPIO_MODE_INPUT, GPIO_PUPD_PULLDOWN, GPIO8);

	/* Configure the EXTI subsystem. */
	exti_select_source(EXTI3, START_BUTTON_PORT);
	exti_direction = RISING;
	exti_set_trigger(EXTI3, EXTI_TRIGGER_RISING);
	exti_enable_request(EXTI3);
}

/* Interrupt Routine*/
void exti3_isr(void){
whatever
}

```
It works great. As expected. When I set it up with EXTI8 instead, it obviously doesn't work either. If I use the lines 
```
nvic_enable_irq(NVIC_EXTI8_IRQ);
void exti8_isr(void)

```
Obviously it throws an error upon build, because there is no `NVIC_EXTI8_IRQ`. If intead, I keep those lines as
```
nvic_enable_irq(NVIC_EXTI3_IRQ);
void exti3_isr(void)
```
But then change the others to
```
gpio_mode_setup(GPIOC, GPIO_MODE_INPUT, GPIO_PUPD_PULLDOWN, GPIO8);

/* Configure the EXTI subsystem. */
exti_select_source(EXTI8, START_BUTTON_PORT);
exti_direction = RISING;
exti_set_trigger(EXTI8, EXTI_TRIGGER_RISING);
exti_enable_request(EXTI8);
```
The code builds and uploads, but the interrupt doesn't ever trigger.

Am I out of luck trying to use pin 8 with libopencm3? Or is there some sort of interrupt handler for pins higher than 4 in the vein of `EXTI9_5_IRQn` that the STM32CubeIDE has?

closed 2020-12-15 by katy-kem

katy-kem, 2020-12-15:
Wow I can't believe this, after typing up this whole thing I realized if I just scrolled down further on the NVIC page there's an `NVIC_EXTI9_5_IRQ`. My bad. Closed.

karlp, 2020-12-16:
Glad we could help :)