libopencm3/libopencm3 #1279: Artefact: overriding a specific source code file from libopencm3 without modifying the library

stoyan-shopov, 2020-12-16:
This is not an issue, and maybe it is a known behavior, but it was not known to me.

I accidentally found that it is possible to override a specific source code file from libopencm3 in a project that uses the library, without actually changing the library source code.

To do so, you can copy the original source code file from the libopencm3 library, and make it a part of your project. Modify the file as needed, but make sure to **NOT** remove any externally visible symbols that the source code file exports. Link as usual. Linking works without giving errors of multiple defined symbols, and the modified source code file takes precedence over the original libopencm3 file. My guess is that, when invoking the linkage editor, the object file for the modified source code file should come before the libopencm3 library.

If some exported, externally visible symbols, are removed from the original file, linking is likely to fail with error for multiply defined symbols.

I found this behavior by accident, I don't know if that is expected behavior, or if it is portable, but for my use case it is very convenient, so I decided to share this experience.

karlp, 2020-12-16:
ok I guess? :) Glad it worked for you?  This will depend a lot on how your own project makefiles/equivalent are set up, it's not really anything that libopencm3 itself controls/enables.  But generally, yes, the linker will look for things it needs in order down the command line.  If you're own project is first, it will override, whether you want that or not.  I don't really think we need to keep this open as an issue though?  I've closed it for now, but let me know if you think there's something that needs to be done here.

closed 2020-12-16 by karlp

stoyan-shopov, 2020-12-16:
No need to keep this open, of course. This behavior just looked curious to me, and was quite useful for my specific use-case, so I just thought it may be worth sharing. This works in the `blackmagic` project, and I also tested an `usart` sample from `libopencm3-examples`. I guess it very much depends on how building the final executable is structured in the project.

Thanks