libopencm3/libopencm3 #1280: Sending of zero-length usb packets

stoyan-shopov, 2020-12-16:
Please, correct me if I am wrong, but the libopencm3 documentation states that `usbd_ep_write_packet()` returns `0 if failed, len if successful`. In some use cases, e.g., usb cdcacm device communication, zero-length usb packets need to be sent, but in that case it looks like that checking the return code of `usbd_ep_write_packet()` is unable to discriminate between success and failure.

I guess changing the API of this function, and breaking everyone's code, is out of the question, so what about adding some function, e.g. `usbd_send_zlp()`, for sending zero-length packets, be added, so that the new function is used in new code?

Thanks

karlp, 2020-12-16:
Yes, I'd agree (from memory) that you can't tell automatically from the return code.   however, you know the length of the data you tried to write, so you should really just be using the check of rc == expected, instead of rc != 0  (This also works for expected == 0!)

So, not a _lot_ of motivation to change the api, but definitely something to consider.  As far as adding a function to explicitly send a zlp, I'd question, strongly, why?! Why would you want to force that different api call onto the library users?  you can see tests for zero length packets in the gadget-zero test cases if you want to look at this further.

stoyan-shopov, 2020-12-16:
Sorry, I am not familiar with the `gadget-zero` testcases...

I will try to explain my concerns.

Here is what documentation for [usbd_ep_write_packet()](http://libopencm3.org/docs/latest/stm32f1/html/group__usb__driver__defines.html#gaa5b245bdd9757cdc79d1fe721ebcf03f) I am looking at.

I don't want a separate function for zero-length packets, but at the moment, `usbd_ep_write_packet()` returns a `uint16_t`. Maybe it would be more appropriate to return some signed type, e.g., an `int`, and have a negative result returned in case of error (e.g., inability to commit a usb packet to the hardware right at the moment). My guess is that changing the result type would break some people's code, that is why I suggested adding a new function. It would definitely be better to change the return type in the API, if the libopencm3 community could tolerate this.

I think there is an issue, specifically for usb cdcacm (virtual serial port devices), where some usb hosts expect that a `short` packet (i.e., of length less than the advertised usb IN endpoint size) is received as a last packet, when 1 or more usb packets of length equal to the advertised usb IN edpoint size are received. This is a peculiarity of the usb cdcacm specification. The last, short packet, which may need to be of length zero, designates to the usb host that the usb transfer is complete, and only then the usb host passes the data upwards the host usb stack. Otherwise, the usb host may wait indefinitely, and not pass the data received to the higher usb layers.

An example, for a usb cdcacm class device: say, your IN endpoint size is 64 bytes. You want to send 70 bytes of data - you send 1 packet of length 64 bytes, and another one, of length `70 - 64 = 6` bytes. The usb host receives the first packet of 64 bytes. As this packet is of length equal to the advertised (64 bytes) usb IN endpoint size, the host does not pass this 64 byte packet upwards the usb stack, but instead waits for more data packets. The host then receives the remaining 6 bytes of data. As this is a `short` packet, i.e., of legth less than the advertised usb IN endpoint size, the usb host assembles a packet of length `64 bytes` (the first packet) `+ 6 bytes` (the second packet) `= 70 bytes` data packet, and sends it upwards the usb stack to whomever awaits this data.

As another example, consider sending a packet of length that is a multiple of the advertised usb data IN endpoint size - for example, a single 64 byte packet. When the host receives the 64 byte packet, it will wait for a `short` packet. But there is no more data to be sent. Therefore, a zero-length packet is necessary as a `short` packet, so that the usb host considers the data transfer complete.

Please, excuse me for being so verbose, I just want to make myself clear...

An obvious workaround, so to say, a `fragmentation` approach, is to never send usb data packets equal to the advertised usb IN endpoint size, so that sending zero-length packets is never necessary. For example, instead of sending a 70 byte packet as `64 + 6 = 70` bytes packet sequence, the same data could be sent as a `63 + 7 = 70` bytes packet sequence. This will work always, but looks like a kind of a botch.

Also, this could probably incur performance penalties - for example, when receiving a packet of length `128 * 64` bytes - the usb host can assemble all `128` packets in one large data packet, and only  then pass this data to the higher usb layers. With the fragmentation approach, when sending a maximum of `64 - 1 = 63` bytes at a time, `(128 * 64) / 63 = 130`, and a last one `(128 * 64) % 63 = 2`, a total of 131 data packets will be transferred - and every single data packet will be sent upwards the usb host stack for processing. In effect - with `fragmentation`, you have 131 data packets thrown upwards the usb host stack, instead of just one packet in the, so to say, `packet assembly` scenario.

Of course, precautions can be made, so that when a large data packet needs to be transferred, it is scheduled that all, but the last data packet are sent as occupying a complete usb IN buffer, and the last transfer, if needed, is split in two `short` packets. But this is maybe getting unnecessarily complicated.

Sorry for the long explanation.

karlp, 2020-12-16:
The api is definitely weak, no argument, but please, a) can you use the check of rc == expected length? (not just rc non-zero) and b) please do look at all the packet size and zero length tests in gadget-zero.

Are you having actual problems that you feel this api change would help you solve? I'm unclear if this is just an "api could be nicer" idea?

stoyan-shopov, 2020-12-17:
Thanks, I will have a look at the usb gadget sources

This is nothing urgent, more like in the spirit of, as you say, api could be nicer. Feel free to close the issue

Thanks

petrutoader, 2021-03-27:
> The api is definitely weak, no argument, but please, a) can you use the check of rc == expected length? (not just rc non-zero) and b) please do look at all the packet size and zero length tests in [...]

In the case of a ZLP, when using `usbd_ep_write_packet()` to send a zero length packet, the return code for a successful write should be `0`, but it's also `0` in the case it fails, right? How am I supposed to know the packet was successfully sent in this case?

stoyan-shopov, 2021-03-30:
I have solved this problem for myself by locally modifying the `libopencm3` library to return to me a value of `0xffff` (because the return type is `uint16_t`) in case of a failure to send the packet

karlp, 2021-03-31:
yeah, after looking at it more, I think the api just definitely needs to be changed.  we're in a "breaking things" phase, so now's as good a time as any.

karlp, 2021-03-31:
@stoyan-shopov  out of interest, how many times did you ever see a failed send of a ZLP?  (I agree the api change is correct, don't worry, just curious whether it happens meaningfully, and.... if could be tested!)

stoyan-shopov, 2021-04-02:
Excuse me for the late reply

I saw a ZLP failure-to-send issiue on the F723 device, if I remember correctly.
That was last year. You are right, it cannot be tested at the moment, I don't have tests to reproduce this right now. I have `gone out of scope`, so to say

Please, let us not close this issue. If I ever manage to reproduce the problem, I will remember to report it

petrutoader, 2021-04-13:
@karlp I could help with modifying the `usbd_ep_write_packet` function, if @stoyan-shopov doesn't have the time to do it. My current project transfers large amounts of data with sizes that are always a power of two. Windows doesn't process the data at all if no ZLP is sent.

stoyan-shopov, 2021-04-14:
Yes, I remember that windows caches CDCACM buffers which are equal to the IN endpoint size, and an applications can get stuck waiting for data. In my last use case that was not such of an issue. In other cases, I worked around by limiting the buffers, which avoided sending a usb packet which is a multiple of the endpoint size, but this hurts performance in high-speed usb usage of libopencm3

@petrutoader What is your idea of fixing the API? Thanks for your effort!

karlp, 2021-04-16:
@petrutoader ZLPs _are_ sent though, this is only an issue about the return code not being able to detect when they _fail_ to send.

If you want to help, figure out how to make a test case following the examples of https://github.com/libopencm3/libopencm3/blob/master/tests/gadget-zero/test_gadget0.py that you can make fail, (and that are valid!) and then a fix.

There's already tests there explicitly for ZLP handling, so work out what's not getting covered and how to fix it.  It's ~trivial to just make the return code negative for failure, but not sure that actually fixes much, you just know that things are now broken :)