libopencm3/libopencm3 #1282: Improved "Getting started" for Windows using WSL2

seymar, 2020-12-16:
Hello,

I really like libopencm3 in comparison to STM32Cube and would love to see more people use it and develop it further.

I remember struggling getting started on Windows, mainly with the prerequisites (msys + OpenOCD). Didn't really like how it worked with msys (it felt dated) and ended up using WSL2 instead. Which I suppose is more common to already have installed these days. There are however some caveats with using OpenOCD from WSL2 accessing USB devices and so.

Is it an idea to change the README or make a getting started using WSL2 with more details on how to set up OpenOCD, GNU make, debugging in VSCode etc.?:
 - WSL2 installation
 - arm-none-eabi on WSL2
 - Python
 - GNU make, building from WSL2 and setting up a VSCode task
 - OpenOCD on Windows for programming
 - arm-none-eabi on Windows for debugging with VSCode with CortexDebug extension

These things all sound very obvious to me now but it would have saved quite a bit of time if I could've followed a more detailed guide.
II may write something out later and turn it into a PR or something but first looking for opinions, experiences and setups of others.

manuelbl, 2020-12-16:
An even easier start is to use Visual Studio Code and [PlatformIO](https://platformio.org/). It requires neither msys nor WSL2. I highly recommend it.

bogdan2011, 2021-02-10:
> An even easier start is to use Visual Studio Code and [PlatformIO](https://platformio.org/). It requires neither msys nor WSL2. I highly recommend it.

Unfortunately platformio doesn't support all the devices that libopencm3 supports.

manuelbl, 2021-02-10:
It's straightforward to add additional devices in PlatformIO. It's probably easier than fiddling with all the Windows workarounds.

Or simply use macOS. It has become the preferred platform of software developers for good reasons.

bogdan2011, 2021-02-10:
I looked for some documentation on adding devices and found nothing.

manuelbl, 2021-02-10:
Fair point. Here's how it's done. I'll demonstrate it for the STM32F042F6. And I'll use Linux/macos notation for directories, in particular `~/.platformio` for PlatformIO's root directory in the users home directory.

1. Go to `~/.platformio/platforms/ststm32/boards` and select the board definition file closest to the MCU you want to add. The chosen file must include a `"libopencm3"` entry.

2. Copy the selected file to `~/.platformio/boards` and rename it to a suitable name.

3. Edit the file and make the necessary changes:

- Add the `"platform": "ststm32"` entry
- Modify all entries with a reference to the MCU name and designation to match your CPU/board
- Set *maximum_ram_size* (RAM) and *maximum_size* (flash) to the correct size

Now you can use the board in your PlatformIO project.

```
{
  "platform": "ststm32",
  "build": {
    "core": "stm32",
    "cpu": "cortex-m0",
    "extra_flags": "-DSTM32F042x6 -DSTM32F042F6",
    "f_cpu": "48000000L",
    "mcu": "stm32f042f6p6"
  },
  "connectivity": [
    "can"
  ],
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32F042F6",
    "openocd_board": "stm32f042f6_stlink",
    "svd_path": "STM32F042x.svd"
  },
  "frameworks": [
    "mbed",
    "stm32cube",
    "libopencm3"
  ],
  "name": "Generic STM32F042F6",
  "upload": {
    "maximum_ram_size": 6144,
    "maximum_size": 32768,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "stlink",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://www.st.com/en/microcontrollers-microprocessors/stm32f042f6.html",
  "vendor": "Generic"
}
```