libopencm3/libopencm3 #1291: stm32l0 select I2C clock source

Spekham2013, 2021-01-04:
Hi!
I am currently writing a program to interface with a stc3100 ic using I2C on a stm32l0. I was looking into to configuring the clock source for the I2C peripheral. I haven't found a function for it yet. I figured it was a dedicated function, but no such luck. My other option was: (libopencm3/stm32/l0/rcc.h)
`void rcc_set_peripheral_clk_sel(uint32_t periph, uint32_t sel);`
But in here I2C1 isn't defined which would cause it to land in the default case and thus not work. Is this a missing feature currently or am I missing something?

Thank a lot!

karlp, 2021-01-04:
did you actually try it?  that is hte function you should use.  I2C1 == I2C1_BASE

karlp, 2021-01-04:
Feel free to re-open this if you are stil having issues.

closed 2021-01-04 by karlp

Spekham2013, 2021-01-04:
Ohh. I see, guess I just missed that. Thanks for the time!