libopencm3/libopencm3 #1296: "Your toolchain doesn't support -mcpu=cortex-m7."Using arm-none-eabi-gcc 9.2.1 on WSL2 ubuntu20.04

Flordsky, 2021-01-23:
It reports me error like the picture when I run make .
[image]
[image]

ChuckM, 2021-01-23:
You have the wrong toolchain installed. Download the toolchain from the arm site and install that. Also, don't build as root, it is unnecessary.

karlp, 2021-01-23:
Actually, just file this as a bug in the project you're building instead.  This project removed that check in September 2018, in cccc1267c5d

closed 2021-01-23 by karlp