libopencm3/libopencm3 #1300: How to handle USB HS device config

UweBonnes, 2021-01-29:
Stoyan Shopov has done a lot of debugging and coding for USB HS with the STM32F723 in the context of libopencm3. It can now be used with as BMP/Firmware  with https://github.com/blacksphere/blackmagic/pull/818. However Usb endpoint wMaxPacketSize and bInterval need to differ depending what speed the device negotiates with the host. Shouldn't get 
''' struct usb_endpoint_descriptor {
...
        uint16_t wMaxPacketSize;
        uint8_t bInterval;
''' get 
'''
        uint16_t wMaxPacketSize[2];
        uint8_t bInterval[2];
'''
at least 2 so we can cope with FS and HS enumeration?  At least 2 , because superspeed enumeration will need another entry, but this will probably be no issue with libopencm3 for some time?

UweBonnes, 2021-01-31:
It it even needed to set wMaxPacketSize  at all? Isn't wMaxPacketSize fixed with the USB speed setting the device has negociated?

UweBonnes, 2021-02-02:
As Zyp pointed out, probably two sets of config descriptors cover different amount of endpoint in FS/HS and different values best.

bm16ton, 2023-05-12:
Sorry for reviving old thread , and also for needing clarification. Im using a homebrew usb-hs libopencm3 driver, and finally after weeks I had the thought. " oh crap what happens when this is plugged into a full speed port?".I assume this is what you are refering to?  I actually have no FS hubs/PC's to test with but I imagine a second set of descriptors for the size and delay are needed. And yes at least on bulk the maxsize is fixed 512. I believe and use smaller then 64 on interrupt endpoints without issue or linux complaining, Trying to find info on 2nd set and best to tell which to send and how brought me here. Anyone with in sights im all ears as they say.