libopencm3/libopencm3 #1302: No clock config for WeAct BlackPill 2.0 (stm32f411xe) 100Mhz

sker65, 2021-02-01:
For this board https://docs.zephyrproject.org/latest/boards/arm/blackpill_f411ce/doc/index.html
there is no clock config for rcc_clock_setup_pll(). The board uses a stm32f411xe and can run up to 100Mhz.

According to CubeMx it should be: 
  RCC_OscInitStruct.PLL.PLLM = 12;
  RCC_OscInitStruct.PLL.PLLN = 96;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 4;

and
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

with flash latency 3.
Can you please provide an additional config. Thx

ChuckM, 2021-02-01:
It is possible to build the clock_scale structure in your code you know this right?

sker65, 2021-02-01:
Yes, of course. Just want to mention that this on eis missing as the board is quite popular these days and not everybody can do the magic clock math P, Q, M, N ....

ChuckM, 2021-02-01:
What I'm saying is that since you're providing documentation, you could also provide example code that sets the clock. The libopencm3 project has gone back and forth at least a couple of times of providing a "higher layer" clock frequency setting API that is more generic (and so doesn't need a custom structure) but the argument against that has been "it has to make too many assumptions and when it doesn't work they will blame the project." 

It is a pretty compelling argument, clocks are one of the more complicated things here. 

That said, you could always submit a pr.

sker65, 2021-02-01:
> That said, you could always submit a pr.
Will do.

closed 2021-02-04 by karlp