libopencm3/libopencm3 #1309: Working USB Examples for STM32F7 (Nucleo-F767ZI)

Poofjunior, 2021-02-07:
Hi there,

I've been working my way through a few examples, but I can't seem to get any USB related examples to work on the STM32F7.

Here's my simplest possible setup, porting [usb_simple_f7.c](https://gist.github.com/Poofjunior/5c3f0ca146dc90c1a8db66df627a8956) originally derived from [usb_simple.c](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/stm32-h107/usb_simple/usb_simple.c) 
I tried to make the fewest number of changes, but the breakdown is:

Use the HSE oscillator with 8MHz input from the attached ST-Link
```
rcc_clock_setup_hse(&rcc_3v3[RCC_CLOCK_3V3_216MHZ], 8);
```
Setup pins:
```
rcc_periph_clock_enable(RCC_GPIOA);
rcc_periph_clock_enable(RCC_OTGFS);

// Relevant Pinouts:
// PA8 --> USB SOF (start of frame, unused)                                                                         
// PA9 --> USB Vbus (otg fs vbus, unused)                                                                           
// PA10 --> USB ID (otg fs id)                                                                                      
// PA11 --> USB DM (Data minus)                                                                                     
// PA12 --> USB DP (Data plus)                                                                                      
// PG6 --> USB GPIO OUT (unused) 
gpio_mode_setup(GPIOA, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO10 | GPIO11 | GPIO12);
gpio_set_af(GPIOA, GPIO_AF10, GPIO10 | GPIO11 | GPIO12);

```
Explicitly disable VBus sensing per [Issue 178](https://github.com/libopencm3/libopencm3-examples/issues/178)
```
OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS | OTG_GCCFG_PWRDWN;
```

Upon uploading the example, I don't get a the expected idVendor/idProduct (0xCAFE, 0xCAFE) show up from running **lsusb**.
Kernel messages show the following:

```
[Feb 7 12:56] usb 1-2.2: new full-speed USB device number 30 using xhci_hcd
[Feb 7 12:57] usb 1-2.2: device descriptor read/64, error -110
[ +15.612012] usb 1-2.2: device descriptor read/64, error -110
[  +0.199754] usb 1-2.2: new full-speed USB device number 31 using xhci_hcd
[ +15.420128] usb 1-2.2: device descriptor read/64, error -110
[ +15.612040] usb 1-2.2: device descriptor read/64, error -110
[  +0.108127] usb 1-2-port2: attempt power cycle
[  +0.627940] usb 1-2.2: new full-speed USB device number 32 using xhci_hcd
[Feb 7 12:58] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[  +5.375970] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[  +0.207981] usb 1-2.2: device not accepting address 32, error -62
[  +0.091994] usb 1-2.2: new full-speed USB device number 33 using xhci_hcd
[  +5.076102] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[  +5.375819] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[  +0.207806] usb 1-2.2: device not accepting address 33, error -62
[  +0.000454] usb 1-2-port2: unable to enumerate USB device
```

For reference, I'm working from a Nucle-f767zi ([databrief](https://www.st.com/resource/en/user_manual/dm00244518-stm32-nucleo144-boards-mb1137-stmicroelectronics.pdf), [datasheet](https://www.st.com/resource/en/datasheet/stm32f767zi.pdf)).

My libopencm3 version is pretty fresh. Working off of master branch updated as of:
```
commit aeb3cee0235dd3e8784c003ff1af82ce44b7b38c (HEAD -> master, origin/master, origin/HEAD)
Author: Karl Palsson <karlp@tweak.net.au>
Date:   Thu Feb 4 10:02:06 2021 +0000
```
It looks like https://github.com/libopencm3/libopencm3/pull/1264 is related but I'm assuming related issues have already been resolved?
Happy to submit a PR into the examples bin for the F7 if I can get this to work!
Cheers--and thanks for taking a look!

manuelbl, 2021-02-08:
Issue #1264 is still pending. So you are most likely affected by it. Chances are you need most of the changes from these pending PRs:

#1258 Fix usb_dwc_common.c endpoint initialization
#1259 Fix usb_dwc_common.c late IN usb interrupt flag clearing
#1261 USB Core 0x2000 fix
#1264 USB OTG support for STM32F7 series

Poofjunior, 2021-02-08:
Oh my goodness--it WORKS! Thanks a bunch for pointing me in the right direction.

Not sure how long these fixes will dangle, so in case anybody stumbles on this later, here's what I did to apply the open pull reqeusts to my local code:

```
cd <your local libopencm3 top directory>
git remote add stoyan-shopov git@github.com:stoyan-shopov/libopencm3.git
git remote add manuelbl git@github.com:manuelbl/libopencm3.git
git fetch stoyan-shopov
git fetch manuelbl

# Create a new branch to merge all the fixes into
git checkout -b stm32f7_usb_fixes

# Merge all the things:
git merge stoyan-shopov/usb-dwc-fix-endpoint-initialization
git merge stoyan-shopov/usb-dwc-fix-late-IN-packet-irq-flag-clearing
git merge manuelbl/no-vbus-sensing
git merge manuelbl/usb-core-0x2000-fix
git merge manuelbl/otg-f7

# recompile libopencm3
make
```

karlp, 2021-02-08:
What would be more useful, is what is the minimum set of those that you need...

Poofjunior, 2021-02-08:
Would you like me to try to cobble together the minimum number of open PRs to get the F7 usb code above working?

UweBonnes, 2021-05-14:
github.com:stoyan-shopov/blackmagic.git stlinkv3 branch has a set of  usb_control.c  usbdfu.c  usb_dwc_common.c  usb_f723.c  files in src/platforms/stlinkv3 that allow to run on the F723

klenSA, 2021-07-10:
Well good :)
Today I am in anticipation that everything will take off and started trying to apply the github.com:stoyan-shopov/blackmagic.git branch.
.. half a day of tinkering with the code ... and ...
it turns out that the usb_f723.c driver uses the HS mode and b14 b15 pins.
It's great!! but what if i need to use FS and a11 a12?

zardam, 2021-08-12:
On a STM32F730, I needed to use "#1261 USB Core 0x2000 fix" to get USB working.