libopencm3/libopencm3 #1310: Multiple EXTIs for pins 5-7? STM32F4

katy-kem, 2021-02-09:
I'm working with an existing board, and the pins I need to set up as external interrupts are PC5, PC6, PC7, PC7, and PA7. Ideally I'd redesign the board and get these signals separated out on pins 0-5 and 10 so I can use seperate external interrupt handlers. However, I'm hoping there's a way to hack it so that multiple pins can trigger the same interrupt routing?

I'm imagining that somehow PC5, PC6, PC7, & PA7 are all set up to trigger the NVIC_EXTI9_5_IRQ handler. Then in the handler I can check the levels on each pin to see what triggered it, and the code runs accordingly. Is this possible? Or am I just SOL and need to redesign the board?

manuelbl, 2021-02-09:
I'm afraid the outlined approach works only partially. You can have EXTI interrupts for PC5, PC6 and PC7 at the same time but not for PC7 and PA7. For each number, you have to select a single port.

For PC5 to PC7, it works along the lines:

```
void setup_exti(void)
{
    nvic_enable_irq(NVIC_EXTI9_5_IRQ);
    
    exti_select_source(EXTI5 | EXTI6 | EXTI7, GPIOC);
    exti_set_trigger(EXTI5 | EXTI6 | EXTI7, EXTI_TRIGGER_BOTH);
    exti_enable_request(EXTI5 | EXTI6 | EXTI7);
}

void exti9_5_isr(void)
{
    if (exti_get_flag_status(EXTI5)) {
        ... PC5 has been triggered, do something ...
        exti_reset_request(EXTI5);
    } else if (exti_get_flag_status(EXTI6)) {
        ... PC6 has been triggered, do something ...
        exti_reset_request(EXTI6);
    } else if (exti_get_flag_status(EXTI7)) {
        ... PC7 has been triggered, do something ...
        exti_reset_request(EXTI7);
    }
}
```

For either PA7 or PC7, you'll have to find an alternative solution.

karlp, 2021-02-10:
I concur, SYSCFG_EXTICRx only lets you enable an single port for each exti, on F4. (g4,l4 and h7 also, I'm not sure if any newer parts have a new exti/syscfg peripheral, or even if it's on the roadmap)

closed 2021-02-10 by karlp