libopencm3/libopencm3 #1312: How right implement sdio driver for libopencm3?

Azq2, 2021-02-14:
I see, in libopencm3 used pattern like "one bit - one function"

I tried to implement SDIO for libopencm3 using this pattern (for sending PR).

Your can see my attempts:
https://github.com/Azq2/libopencm3/blob/13f87bda66de0dbd76b078d55e0fe27e269a7177/include/libopencm3/stm32/common/sdio_common_all.h
https://github.com/Azq2/libopencm3/blob/13f87bda66de0dbd76b078d55e0fe27e269a7177/lib/stm32/common/sdio_common_all.c

But when i tried to use this i found unsolvable problem. STM32 used modified AMBA PL180 peripheral. And its regs very slow. 
And on fast CPU's (in my case - 186 MHz) i have this problems:

For example:
```C
// NOT WORK
SDIO_CMD(sdio) &= ~SDIO_CMD_CPSMEN;
SDIO_CMD(sdio) |= SDIO_CMD_CPSMEN;

if (SDIO_CMD(sdio) & SDIO_CMD_CPSMEN) { // false!!!!
	// this code not run :(
}

// WORKS FINE
SDIO_CMD(sdio) &= ~SDIO_CMD_CPSMEN;
delay_ms(1);

SDIO_CMD(sdio) |= SDIO_CMD_CPSMEN;
delay_ms(1);

if (SDIO_CMD(sdio) & SDIO_CMD_CPSMEN) { // true, as expected
	// this code runs!
}
```

What we see - after writing to the register, any reading from it for some time returns its old value.

Because of this, it becomes impossible to use partial update of bits in registers.
There is only one way - writing the full value to the register, without reading the old one.

Like this:
```C
uint32_t tmpreg = 0;

if (wait_response) {
	tmpreg |= SDIO_CMD_WAITRESP;
} else {
	tmpreg &= ~SDIO_CMD_WAITRESP;
}

if (long_response) {
	tmpreg |= SDIO_CMD_LONGRESP;
} else {
	tmpreg &= ~SDIO_CMD_LONGRESP;
}

tmpreg |= SDIO_CMD_CPSMEN;
tmpreg |= index;

SDIO_CMD(sdio) = tmpreg;
```

This way used in HAL or LL.

I tested all other SDIO register.... and i have bad news. All important registers are slow.

My code for testing:
```C

int test_reg_for_slowpoke(uint32_t reg, uint8_t bit) {
	uint32_t old = MMIO32(reg);
	
	MMIO32(reg) &= ~(1 << bit);
	MMIO32(reg) |= (1 << bit);
	
	return !(MMIO32(reg) & (1 << bit));
}

int test_reg_for_wr(uint32_t reg, uint8_t bit) {
	uint32_t old = MMIO32(reg);
	
	MMIO32(reg) |= (1 << bit);
	delay_ms(2);
	bool a = (MMIO32(reg) & (1 << bit));
	
	MMIO32(reg) &= ~(1 << bit);
	delay_ms(2);
	bool b = (MMIO32(reg) & (1 << bit));
	
	return a != b;
}

void test_reg(const char *name, uint32_t reg) {
	printf("%s - 0x%08lX:\r\n", name, reg);
	for (int i = 0; i < 32; i++) {
		bool is_slow = false;
		
		for (int j = 0; j < 100; j++) {
			if (test_reg_for_slowpoke(reg, i))
				is_slow = true;
		}
		
		bool is_wr = test_reg_for_wr(reg, i);
		
		printf("  -> bit %d ", i);
		
		if (is_wr) {
			printf(" [wr]");
		} else {
			printf(" [ro]");
		}
		
		if (is_slow && is_wr)
			printf(" [slow]");
		
		printf("\r\n");
	}
	
}

int sdio_test(void) {
	test_reg("SDIO_POWER", SDIO1 + 0x00);
	test_reg("SDIO_CLKCR", SDIO1 + 0x04);
	test_reg("SDIO_ARG", SDIO1 + 0x08);
	test_reg("SDIO_CMD", SDIO1 + 0x0C);
	test_reg("SDIO_RESPCMD", SDIO1 + 0x10);
	test_reg("SDIO_RESP1", SDIO1 + 0x14);
	test_reg("SDIO_RESP2", SDIO1 + 0x18);
	test_reg("SDIO_RESP3", SDIO1 + 0x1C);
	test_reg("SDIO_RESP4", SDIO1 + 0x20);
	test_reg("SDIO_DTIMER", SDIO1 + 0x24);
	test_reg("SDIO_DLEN", SDIO1 + 0x28);
	test_reg("SDIO_DCTRL", SDIO1 + 0x2C);
	test_reg("SDIO_DCOUNT", SDIO1 + 0x30);
	test_reg("SDIO_STA", SDIO1 + 0x34);
	test_reg("SDIO_ICR", SDIO1 + 0x38);
	test_reg("SDIO_MASK", SDIO1 + 0x3C);
	test_reg("SDIO_FIFOCNT", SDIO1 + 0x48);
	test_reg("SDIO_FIFO", SDIO1 + 0x80);
}
```

And its sad output: https://gist.github.com/Azq2/90fc4b136e7c523392e47676e6cd01be

And yes, `test_reg_for_slowpoke` function works right:
```asm
34: sym.test_reg_for_slowpoke (int16_t arg1, int16_t arg2);
; arg int16_t arg1 @ r0
; arg int16_t arg2 @ r1
0x08000330      ldr     r3, [r0]   ; arg1
0x08000332      ldr     r2, [r0]   ; arg1
0x08000334      movs    r3, 1
0x08000336      lsl.w   r1, r3, r1
0x0800033a      bic.w   r2, r2, r1 ; arg2
0x0800033e      str     r2, [r0]   ; arg1
0x08000340      ldr     r3, [r0]
0x08000342      orrs    r3, r1
0x08000344      str     r3, [r0]
0x08000346      ldr     r3, [r0]
0x08000348      tst     r1, r3
0x0800034a      ite     eq
0x0800034c      movs    r0, 1
0x0800034e      movs    r0, 0
0x08000350      bx      lr
```

There are several API ways for implementing this method:

### 1. One complex function per one register and structures in function arguments:
```C
sdio_clock_init_t clock_init = {
	enable			= true,
	flow_control	= true,
	clock_phase		= SDIO_CLOCK_PHASE_0,
	bus_width		= SDIO_CLKCR_WIDBUS_1,
	bypass			= false,
	power_save		= true,
	div				= 118
};
sdio_clock_init(&clock_init);
```

### 2. One complex function per one register and OVER_9999 arguments in function:
```C
// what the hell is going on here...
sdio_clock_init(true, true, SDIO_CLOCK_PHASE_0, SDIO_CLKCR_WIDBUS_1, false, true, 118);
```

### 3. Manual register value buffering v1
```C
uint32 clock_reg;
sdio_clock_reg_init(&clock_reg); // readback reg from hw
sdio_clock_reg_enable(&clock_reg);
sdio_clock_reg_enable_flow_control(&clock_reg);
sdio_clock_reg_enable_bypass(&clock_reg);
sdio_clock_reg_set_bus_width(&clock_reg, SDIO_CLKCR_WIDBUS_1);
sdio_clock_reg_commit(&clock_reg); // writeback reg to hw
```

### 4. Manual register value buffering v2
```C
sdio_clock_reg_sync(sdio); // optional readback reg from hw

sdio_set_clock_phase_0(sdio); // update internal buffered value
sdio_disable_clock_bypass(sdio); // update internal buffered value
sdio_set_clock_div(sdio, SDIO_CLK_400KHZ); // update internal buffered value
sdio_disable_power_save(sdio); // update internal buffered value
sdio_set_bus_width(sdio, SDIO_CLKCR_WIDBUS_1); // update internal buffered value
sdio_disable_flow_control(sdio); // update internal buffered value

sdio_clock_reg_commit(sdio); // writeback reg to hw
```

All this ways not enought good and i dont know how it right implement in libopencm3. 

Your ideas? How do it right?

Azq2, 2021-02-14:
Or, add `DSB` to all functions? Crazy idea.

karlp, 2021-02-14:
dsb is probably the only correct path. I'm not sure I follow what you're numbered 1..4 examples are trying to convey though.

Azq2, 2021-02-15:
> dsb is probably the only correct path. I'm not sure I follow what you're numbered 1..4 examples are trying to convey though.

DSB not works. :(

After detailed reading reference manual i found this line:
```
The adapter registers and FIFO use the APB2 bus clock domain (PCLK2). The control unit,
command path and data path use the SDIO adapter clock domain (SDIOCLK).
```
It means:
- The microcontroller accesses the SDIO block using PCLK2.
- SDIO block updates its **internal** registers using SDIOCLK
- `DSB` syncs PCLK2 but not SDIOCLK. The SDIO block is a black box for the CPU.

As a result: SDIO block and APB2 are out of sync.

Because of this, we get a lag between updating the register from the MCU and actually updating it inside the SDIO block.

At F4, this delay is:
```
count_of_nops = PCLK2 / SDIOCLK
```

If add `count_of_nops` NOPs before any read or after any write of register, then everything works perfectly.

Also, if set PCLK equal to SDIOCLK, everything works fine. 

This way works at any CPU or PCLK2 frequency.

But who should take care of this delay after every register update. User or library?

At now, there is no SDIOCLK data in the library and it is impossible to calculate the required delay for synchronization.

And I think the delay in each function will create a serious performance problem.

At the moment, the most reliable option remains - to update the entire register, instead of updating individual bits. In this case, reading the old value is not required and we always get exactly what we want to write to it.

But this entails creating complex functions, sometimes with >10 arguments. This might be inappropriate for inclusion in this library.

closed 2021-05-15 by Azq2