libopencm3/libopencm3 #1318: stm32/f4/nvic.h file not found

denizzzka, 2021-03-03:
nvic.h and vector_nvic.c is missed for stm32f4?

(I use my own build script, `-DSTM32F4` option is passed to the compiler)

```
In file included from ../subprojects/libopencm3/lib/cm3/nvic.c:46:
In file included from ../subprojects/libopencm3/include/libopencm3/cm3/nvic.h:147:
../subprojects/libopencm3/include/libopencm3/dispatch/nvic.h:14:11: fatal error: 'libopencm3/stm32/f4/nvic.h' file not found
#       include <libopencm3/stm32/f4/nvic.h>
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

When I dirty-fix this by substituting path to f3 version instead, another error also occured:
```
In file included from ../subprojects/libopencm3/lib/cm3/vector.c:28:
../subprojects/libopencm3/lib/cm3/../dispatch/vector_nvic.c:10:11: fatal error: '../stm32/f4/vector_nvic.c' file not found
#       include "../stm32/f4/vector_nvic.c"
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

denizzzka, 2021-03-03:
Ah, probably here is autogeneration should be used

closed 2021-03-03 by denizzzka

karlp, 2021-03-03:
correct :) if you're using your own build scripts, you're on your own :)