libopencm3/libopencm3 #1321: Linux compiling issue caused by \ in end of lib name

devidasjadhav, 2021-03-28:
https://github.com/libopencm3/libopencm3/blob/3922cc7d3ea362b816cc1de6d3287375afbe58aa/Makefile#L28

diff --git a/Makefile b/Makefile
index 0527e3e2..0af2f426 100644
--- a/Makefile
+++ b/Makefile
@@ -25,7 +25,7 @@ STYLECHECKFLAGS := --no-tree -f --terse --mailback
 
 space:=
 space+=
-SRCLIBDIR:= $(subst $(space),\$(space),$(realpath lib))
+SRCLIBDIR:= $(subst $(space),/$(space),$(realpath lib))
 
 TARGETS := stm32/f0 stm32/f1 stm32/f2 stm32/f3 stm32/f4 stm32/f7
 TARGETS += stm32/l0 stm32/l1 stm32/l4
fixes the issue.

before raising pull request can anyone confirm windows compatibility with my patch?
without my patch Make file creating make command

`make --directory=lib/stm32/f0 SRCLIBDIR="/home/dev/Downloads/git/stm32/libopencm3/lib\"`

error message with extra logs printing variables.
```
$ make            
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  BUILD   lib/stm32/f0
LIB_DIRS=lib/stm32/f0 lib/stm32/f1 lib/stm32/f2 lib/stm32/f3 lib/stm32/f4 lib/stm32/f7 lib/stm32/l0 lib/stm32/l1 lib/stm32/l4 lib/lpc13xx lib/lpc17xx lib/lpc43xx/m4 lib/lpc43xx/m0 lib/lm3s lib/lm4f lib/efm32/tg lib/efm32/g lib/efm32/lg lib/efm32/gg lib/sam/3a lib/sam/3n lib/sam/3s lib/sam/3u lib/sam/3x lib/sam/d lib/vf6xx
SRCLIB_DIRS=/home/dev/Downloads/git/stm32/libopencm3/lib
/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/bin/sh: -c: line 2: syntax error: unexpected end of file
make: *** [Makefile:70: lib/stm32/f0] Error 2
```

karlp, 2021-03-28:
what?  did you run, 
```
make --directory=lib/stm32/f0 SRCLIBDIR="/home/dev/Downloads/git/stm32/libopencm3/lib\"
```
Where did you get this? what platform are you on? Nothing prints out like that out of the box, do you have other local modifications?

The code you're "patching" doesn't even exist, it was removed in 72274d3caf09 and was never used anyway, as all targets specified SRCLIBDIR themselves.

If you can demonstrate how you managed this with unmodified code, feel free to re-open it again, but I'm closing it for now.

closed 2021-03-28 by karlp

devidasjadhav, 2021-03-29:
@karlp you are right.
the problem was with repo I was using
https://github.com/ve3wwg/stm32f103c8t6

which had this at older head  3922cc7d3ea362b816cc1de6d3287375afbe58aa. 
which caused this.
Apologies for this mistake.