libopencm3/libopencm3 #1322: STM32F1, USB CDC, issues with writing to bulk endpoint.

krish2487, 2021-04-03:
Hello folks,
I am working with a STM32F103RCT6 and trying to implement USB CDC on the same processor. Before I get into the bulk of the problem, a little bit of peripheral information which I think is relevant. The app is structured in layers.

 

- I have a file app.c which has the main function. 
-  main() calls the Init() (present in Init.c) which brings up the board - POST, GPIO setup, peripheral setup and so on. 
-  Init calls LogInit and LogMessage functions (present in Log.c) as part of the bring up sequence. 
-  LogInit calls InitUSB ( present in DriverUSB.c ) and LogMessage calls WriteUSBMessage ( DriverUSB.c ).
-  DriverUSB.c sits on top of LibopenCM3 HAL and calls the necessary functions to initialize the USB peripheral.

Whew, now the problem. The board does not enumerate on the PC. 
Please find the file contents of the aforementioned files.

app.c 
```
#include "app.h"
#include "BSP.h"
#include "Init.h"

#ifndef TEST
	// #define MAIN main
	int main(void)
#else
	// #define MAIN testable_main
	int testable_main(void)
#endif

// int MAIN(int argc, char const *argv[])
{

	Init();

	#ifndef TEST
	while (1)
		{

		}
	#else
	#endif
	/* code */
	return 0;
}

```

Init.c
```
#include "Init.h"
#include "IO.h"
#include "Log.h"

enum Error_codes Init(void)
{

	SysClockInit();
	if (ERROR == IOInit(SYS_STATUS))
		return ERROR;
	IOBlink(POWERUP);
	LogInit(USB);
	LogMessage(USB, INFO_LEVEL, "Initializing System");

	return SUCCESS;
}

```

Log.c

```
[... 343 lines cut ...]
Now the problem I am seeing is here, in the Init.c file.

```
	LogInit(USB);
	LogMessage(USB, INFO_LEVEL, "Initializing System");
```

The board enumerates perfectly fine and I can see a USB CDC device  when I comment out
```
	LogMessage(USB, INFO_LEVEL, "Initializing System");
```

but fails, when I add it. I have tried debugging the code with Ozone and Jlink edu. And it looks like the LogMessage causes  a HardFault Exception.

This is what the call stack looks like 
**reset_handler -> main -> Init -> LogMessage -> WriteUSBMessage -> st_usbfs_ep_write_packet -> st_usbfs_copy_to_pm -> blocking_handler**

st_usbfs_ep_write_packet is present in st_usbfs_core.c
st_usbfs_copy_to_pm is present in st_usbfs_v1.c

AFAICS, something in the definition of my LogMessage is causing the crash in the copy_to_pm function, I suspect it is something about the length of the message.. but strlen should just return the length of the string, I am attempting to write. 

I will be grateful if someone more knowledgeable about the USB Stack implementation can point out what I am doing wrong. Hopefully it is a silly issue from my code.. 

Thanks in advance folks!! and have a wonderful weekend!! :-)

manuelbl, 2021-04-03:
Can you better describe the behavior you are observing? You write that the device does not enumerate on the PC. Does it mean that the USB device does not appear at all (in the Windows Device Manager) or does it mean that it does appear but not as a COM port?

I've seen that the code toggles a GPIO pin in the USB interrupt handler? How many times does it toggle when you plug it in?

The STM32 F103 series does not have a built-in pull-up resistor on D+. Can you give more details about your hardware and how you have implemented the pull-up resistor?

krish2487, 2021-04-03:
@manuelbl  
Thank you for the prompt response. The device does not enumerate at all, when the "LogMessage" function is uncommented. 
When I do comment it out, the device shows up as a USB CDC device and I can even connect using a terminal emulator, however it does not print anything. 

When the LogMessage is commented out, the LED toggles, when I connect or disconnect the terminal emulator from the port. 
When the LogMessage is uncommented, the LED does not toggle at all. 

It is a custom board that I have designed and the hardware has been validated quite a while ago. 
[image]

As you can see R204 is the pull up on the D+ line. IIRC it was a 4K7 pull up.

manuelbl, 2021-04-03:
I might have spotted the problem. The proper sequence of initialization (related to USB) is:

1. Call `usbd_init()`
2. Register a configuration callback using `usbd_register_set_config_callback()`
3. When the configuration callback is called, create endpoints using `usbd_ep_setup()`
 
Only after step 3 has been completed, `LogMessage` may be used. Before that, it's likely to crash.

As far as I can see, `usbd_register_set_config_callback()` is not called at all. Thus the configuration callback `cdcacm_set_config()` isn't called either and no endpoints are ever set up.

The below changes should fix the problem:

DriverUSB.h
```
extern int is_usb_configured;
```

DriverUSB.c
```
int is_usb_configured = 0;

enum Error_codes InitUSB(void)
{
    ...
    nvic_enable_irq(NVIC_USB_WAKEUP_IRQ);
    usbd_register_set_config_callback(dev, cdcacm_set_config);
    return SUCCESS;
}

void cdcacm_set_config(usbd_device *usbd_dev, uint16_t wValue)
{
    ...
    usbd_register_control_callback(
        usbd_dev,
        USB_REQ_TYPE_CLASS | USB_REQ_TYPE_INTERFACE,
        USB_REQ_TYPE_TYPE | USB_REQ_TYPE_RECIPIENT,
        cdcacm_control_request);
    is_usb_configured = 1;
}
```

Log.c
```
enum Error_codes LogMessage(enum LogSubSystem System, enum LogLevels level, char *message)
{
    if (!is_usb_configured)
        return ERROR;

    WriteUSBMessage(message);
    return SUCCESS;
}
```

krish2487, 2021-04-03:
@manuelbl 
You are right.. I totally missed calling the usbd_register_set_config_callback function.. 
🤦 
I m reasonably confident that this should fix the issue.. Please do give me some time to check it on the hardware. I am a little away from the hardware atm. 

Nevertheless, thank you for the help and the quick eyes to spot the issue. :-)

krish2487, 2021-04-03:
Well it does sort of work... and also doesnt.
You were right about the usbd_register_set_config_callback not being called. I did make the changes you suggested and this function is called.

However, the cdcacm_set_config function is never called. I.e the endpoints are never setup. and there is nothign printed on the serial port since it returns an error. I did step and verify that the cdcacm_set_config is never called. it just returns from the register_set_config_callback function. 

There is one change I did to the above mentioned changes, you have suggested.
You have the cdcacm_set_config defined after the InitUSB function. I moved it before the InitUSB function, i.e as in the original source file I posted above.

krish2487, 2021-04-03:
Ok.. a little more debugging..
I modified the InitUSB function to so

```
enum Error_codes InitUSB(void)
{

	rcc_periph_clock_enable(RCC_GPIOA);
	rcc_periph_clock_enable(RCC_USB);
	rcc_periph_reset_pulse(RST_USB);
	usbd_dev =  usbd_init(&st_usbfs_v1_usb_driver, &dev, &config, usb_strings, 3, usbd_control_buffer, sizeof(usbd_control_buffer));
	usbd_register_set_config_callback(usbd_dev, cdcacm_set_config);
	nvic_set_priority(NVIC_USB_LP_CAN_RX0_IRQ, 2 << 6);
	nvic_enable_irq(NVIC_USB_LP_CAN_RX0_IRQ);
	nvic_enable_irq(NVIC_USB_WAKEUP_IRQ);
	if (is_usb_configured == 0)
		return ERROR;
	return SUCCESS;
}
```
to check if the cdcacm_set_config callback is actually being called and the Endpoints are being set.
The function returns ERROR. meaning that the 
```
is_usb_configured
```
variable is never being set. This makes me suspect that the cdcacm_set_config is not being called at all, since that is the only place we are actually setting the variable. I stepped thru the code with the debugger and can confirm this.

@manuelbl  Any suggestions on what I can try next?

manuelbl, 2021-04-03:
You seem to expect that `cdcacm_set_config()` is immediately called from `usbd_register_set_config_callback()`. That's not the case.  `cdcacm_set_config()` will be called once Windows decides to configure the USB device.

Therefore, you will not be able to send any log messages right after startup. You should probably remove the statement `LogMessage(USB, INFO_LEVEL, "Initializing System");` as it cannot possibly work. Log output that is generated later (probably about 2s after you have plugged in the device) will be successfully sent.

So what's the overall behavior now? Does the device appear as a COM port in Windows but not generate any output? Or does it no longer appear as a COM port? I hope that at least the hard fault has been fixed.

krish2487, 2021-04-03:
Oh my mistake, I assumed that it would be called immediately. I didnt realise it would depend on the host system to actually instantiate the endpoints.

That is correct. The device shows up as a COM port but does not send any messages. 
Would introducing a delay of about 5 seconds give it enough time to enumerate?? 
between the set_config_callback and the cdcacm_set_config being called?? Less elegant but I guess I can refactor it into the code into something more practical if such a solution works...

krish2487, 2021-04-03:
I can confirm that introducing a delay of about 5 seconds between the InitUSB and LogMessage indeed does allow the host to configure the device correctly and the initial message is printed to the console as well.

@manuelbl  I owe a thanks and beer to you for being so astute and being able to pinpoint the source of the problem at hand - me . :-)

Thank you again fro your help and suggestions!! I am going to close this issue now and continue on my travels.. :-P

closed 2021-04-03 by krish2487