libopencm3/libopencm3 #1331: Issue relaying messages between multiple USART connections on the STM32F1 Blue pill

Mathiasbjensen, 2021-04-27:
I am using the STM32F1 Blue Pill, which is connected to a SARA-R412 module on USART2 and an ESP8266 is connected on USART3. 

I would like the STM32 to send a simple AT command to the SARA module, then receive a response from the SARA module and relay it to the ESP. 

Currently I am able to correctly relay a couple of messages, but then after that gibberish is just relayed, either  there's a lot of missing characters in the message relayed to the ESP, or it's just the same character being transmitted. 

I have verified with TTL that the message is correctly sent from the SARA module and that the STM receives it correctly. The issue seems to be on the STM32 with the way I handle the message.

I have made a fairly simple `get_string` method for receiving a message on 1 UART port and relaying it to another UART port.

```c
static void usart_get_string(uint32_t usartSrc, uint32_t usartDst, uint16_t str_max_size)
{
    uint8_t received = 'V';
    uint16_t itr = 0;
    uint8_t recvPrev;

    while (itr < str_max_size)
    {
        if (!((USART_SR(USART2) & USART_SR_RXNE) == 0))
        {
            received = usart_recv_blocking(usartSrc);	
        }
        uart_putc(received, usartDst);

        if (recvPrev == 'O' && received == 'K')
        {
            break;
        }
        recvPrev = received;
        uart_putc_blocking(itr, usartDst); // Somehow doesn't work at all without this
        itr++;
    }
}
```
The main flow of the program of the STM32 is just a simple loop which sends an AT command to the SARA module and then calls `get_string()`.

While testing I omitted the ESP for now and just read the STM's output via TTL.
I also made a [Stackoverflow Post](https://stackoverflow.com/questions/67215309/issue-relaying-messages-between-multiple-usart-connections-on-the-stm32f1-blue-p) which contains full code and some pictures of setup/output.

reymor, 2021-07-05:
Hi! Have you solved this problem?

neoxic, 2022-01-12:
First, you don't need to wait for RXNE yourself:
```
if (!((USART_SR(USART2) & USART_SR_RXNE) == 0))
```
since `usart_recv_blocking()` does the job for you.

Second, it's hard to tell what your code does without any insight into `uart_putc()` and `uart_putc_blocking()`.