libopencm3/libopencm3 #1340: STM32L432 usb device

codespotx, 2021-05-23:
hi guys, hi Karl,

i currently struggle with creating a functional usb hid device with the STM32L432 using the internal clock.
there is no specific example so i "ported" the according STM32F1 example (hidusb.c), basically by switching to "st_usbfs_v2_usb_driver", and playing with clock/crs options. 

the result is always a "dead" device. linux kernel shows error when reading the descriptor ("device descriptor read/64, error -71").
the hardware should be OK, in USB DFU mode the according device shows up and works correctly.

[usb_l432.zip](https://github.com/libopencm3/libopencm3/files/6527878/usb_l432.zip)

any hint or working sample would be helpful.

manuelbl, 2021-05-23:
The STM32L432 isn't well supported when it comes to initialize the clock for USB. I've successfully solved it in the past (see the code below).

The code configures the clocks as shown in this screenshot from STM32cubeMX, i.e. setting an internal clock of 80 MHz and no external high-speed clock. Other configurations are possible. Ignore the "PPLSAI1" part as it is not configured.

[image]

It looks as if there are still a few bits missing in your code. By replacing the clock initialization code, I can get your code to work and the USB device to appear. It is also visible with `dfu-util`.

> USB DFU mode the according device shows up and works correctly

If you are using a Nucelo-L432KC, then what you are seeing is most likely the ST-Link on the Nucleo board. This wouldn't say much about the separate STM32L432 CPU itself. On a Nucelo-L432KC board, the uploaded code uses a separate USB port on pins PA11 and PA12. You would have to fit a USB connector to it.

Regarding the *Makefile*: I didn't use yours as it depends on other stuff not included in the zip file. Thus I couldn't verify if it works. I don't understand either how to properly use the ones provided on GitHub. None seems to be suitable for a small standalone project. So I've created a *PlatformIO* project that's easy to use: [usbhid-STM32L432.zip](https://github.com/libopencm3/libopencm3/files/6528712/usbhid-STM32L432.zip)

```c
    /* Initial setup: MSI active, 4 MHz */

    /* Temporarily enable internal high-speed oscillator (HSI). */
    rcc_osc_on(RCC_HSI16);
    rcc_wait_for_osc_ready(RCC_HSI16);

    /* Select HSI as SYSCLK source. */
    rcc_set_sysclk_source(RCC_CFGR_SW_HSI16);

    /* Set the VOS scale mode */
    rcc_periph_clock_enable(RCC_PWR);
    pwr_set_vos_scale(PWR_SCALE1);

    /* Enable external low-speed (LSE) clock for auto-trimming MSI clock. */
    pwr_disable_backup_domain_write_protect();
    rcc_osc_on(RCC_LSE);
    rcc_wait_for_osc_ready(RCC_LSE);
    pwr_enable_backup_domain_write_protect();

    /* Enable multispeed internal oscillator (MSI). */
    rcc_osc_on(RCC_MSI);
    rcc_wait_for_osc_ready(RCC_MSI);

    /* Increase MSI clock ro 48 MHz. */
    rcc_set_msi_range(RCC_CR_MSIRANGE_48MHZ);

    /* Set prescalers for AHB, ADC, APB1, APB2.*/
    rcc_set_hpre(RCC_CFGR_HPRE_NODIV);
    rcc_set_ppre1(RCC_CFGR_PPRE1_NODIV);
    rcc_set_ppre2(RCC_CFGR_PPRE2_NODIV);

    /* Disable PLL oscillator before changing its configuration. */
    rcc_osc_off(RCC_PLL);

    /* Configure the main PLL */
    rcc_set_main_pll(RCC_PLLCFGR_PLLSRC_MSI, 3, 10, RCC_PLLCFGR_PLLP_DIV7, RCC_PLLCFGR_PLLQ_DIV2, RCC_PLLCFGR_PLLR_DIV2);

    /* Enable PLL oscillator and wait for it to stabilize. */
    rcc_osc_on(RCC_PLL);
    rcc_wait_for_osc_ready(RCC_PLL);

    /* Slower flash settings for higher clock speed. */
    flash_dcache_enable();
    flash_icache_enable();
    flash_set_ws(FLASH_ACR_LATENCY_4WS);

    /* Select PLL as SYSCLK source. */
    rcc_set_sysclk_source(RCC_CFGR_SW_PLL);

    /* Wait for PLL clock to be selected. */
    rcc_wait_for_sysclk_status(RCC_PLL);

    /* Set the peripheral clock frequencies used. */
    rcc_ahb_frequency  = 80000000;
    rcc_apb1_frequency = 80000000;
    rcc_apb2_frequency = 80000000;

    /* Disable internal high-speed oscillator. */
    rcc_osc_off(RCC_HSI16);

    /* Select MSI as 48MHz clock source */
    rcc_set_clock48_source(RCC_CCIPR_CLK48SEL_MSI);

    /* Enable the MSI clock PLL (derived from LSE) */
    RCC_CR |= RCC_CR_MSIPLLEN;

    /* Enable USB supply */
    PWR_CR2 |= PWR_CR2_USV;
```

codespotx, 2021-05-26:
Hi Manuel,

thank you for the fast and detailed response, unfortunately the code does not work for me.

about my hardware:
I use a minimalistic crystal less board with stm32l432kc, USB is connected to PA11 and PA12.
programming via ST-Link and openocd.

about the Makefile:
the build is based on https://github.com/libopencm3/libopencm3-template
(did not touch "rules.mk", this file is missing in the zip)

So without the LSE a different clock setup is required, will check all the code snippets again.

manuelbl, 2021-05-26:
Here's the modified clock initialization code that should work without crystal:

```
    /* Initial setup: MSI active, 4 MHz */

    /* Temporarily enable internal high-speed oscillator (HSI). */
    rcc_osc_on(RCC_HSI16);
    rcc_wait_for_osc_ready(RCC_HSI16);

    /* Select HSI as SYSCLK source. */
    rcc_set_sysclk_source(RCC_CFGR_SW_HSI16);

    /* Set the VOS scale mode */
    rcc_periph_clock_enable(RCC_PWR);
    pwr_set_vos_scale(PWR_SCALE1);

    /* Enable multispeed internal oscillator (MSI). */
    rcc_osc_on(RCC_MSI);
    rcc_wait_for_osc_ready(RCC_MSI);

    rcc_osc_on(RCC_HSI48);
    rcc_wait_for_osc_ready(RCC_HSI48);

    /* Increase MSI clock ro 48 MHz. */
    rcc_set_msi_range(RCC_CR_MSIRANGE_48MHZ);

    /* Set prescalers for AHB, ADC, APB1, APB2.*/
    rcc_set_hpre(RCC_CFGR_HPRE_NODIV);
    rcc_set_ppre1(RCC_CFGR_PPRE1_NODIV);
    rcc_set_ppre2(RCC_CFGR_PPRE2_NODIV);

    /* Disable PLL oscillator before changing its configuration. */
    rcc_osc_off(RCC_PLL);

    /* Configure the main PLL */
    rcc_set_main_pll(RCC_PLLCFGR_PLLSRC_MSI, 3, 10, RCC_PLLCFGR_PLLP_DIV7, RCC_PLLCFGR_PLLQ_DIV2, RCC_PLLCFGR_PLLR_DIV2);

    /* Enable PLL oscillator and wait for it to stabilize. */
    rcc_osc_on(RCC_PLL);
    rcc_wait_for_osc_ready(RCC_PLL);

    /* Slower flash settings for higher clock speed. */
    flash_dcache_enable();
    flash_icache_enable();
    flash_set_ws(FLASH_ACR_LATENCY_4WS);

    /* Select PLL as SYSCLK source. */
    rcc_set_sysclk_source(RCC_CFGR_SW_PLL);

    /* Wait for PLL clock to be selected. */
    rcc_wait_for_sysclk_status(RCC_PLL);

    /* Set the peripheral clock frequencies used. */
    rcc_ahb_frequency  = 80000000;
    rcc_apb1_frequency = 80000000;
    rcc_apb2_frequency = 80000000;

    /* Disable internal high-speed oscillator. */
    rcc_osc_off(RCC_HSI16);

    crs_autotrim_usb_enable();

    /* Select HSI48 as 48MHz clock source */
    rcc_set_clock48_source(RCC_CCIPR_CLK48SEL_HSI48);

    /* Enable USB supply */
    PWR_CR2 |= PWR_CR2_USV;
```

And this is the configuration it's supposed to implement:

[image]

I've tested it with the PlatformIO project and with a Nucelo board with crystal. So I can't really verify it works without crystal.

codespotx, 2021-05-27:
Hi Manuel,

this code is working now with my hardware.

thank you for your support

closed 2021-05-27 by codespotx