libopencm3/libopencm3 #1354: License issues

doragasu, 2021-08-09:
First of all, sorry for bringing this topic back.

I know this has been discussed in the past, and I suppose some devs might not be comfortable discussing this topic. But other than the info in the [Licensing wiki page](https://github.com/libopencm3/libopencm3/wiki/Licensing) and some brief posts in the repository issues, [the last discussion I found](https://sourceforge.net/p/libopencm3/mailman/libopencm3-devel/thread/D53F4B84-7C98-473A-A871-E03DD29F27D8%40esden.net/#msg30574571) was from 2013, and it seems most devs were OK to accept the linking exception.

I am a I free software advocate, but I think using LGPL without linking exception for embedded libraries that cannot be dynamically linked is both confusing for developers and users, and can cause more harm than benefit. Not having the linking exception can scare many devs, can cause lots of troubles mixing libopencm3 with other incompatible licensed libraries and has, in my opinion, almost negligible benefits.

I personally am OK with all other LGPL "restrictions", including the anti tivoization clauses. But please reconsider explicitly adding the linking exception to the license, so devs using the library are still forced to contribute new features and bugfixes, but can use it without fear of "contaminating" other libraries they use in their projects, or being forced to releasing source that is not related to libopencm3.

Thanks for the library, and again, sorry to bother with this topic, but I think it is important.

// doragasu

karlp, 2021-08-10:
There is absolutely no feasible way of relicensing this project at this point in time given the number of contributors that would have to be tracked down, regardless of the wishes of some contributors in either direction.  LGPL was/is already a compromise, some contributors want(ed) GPL, some want(ed) LGPL+linking exception and/or more permissively.  For better or worse, it's now pretty firmly stuck right where it is now on LGPL.

karlp, 2021-08-10:
It's worth mentioning that you are not required to dynamically link, nor are you required to release unrelated source.  You are required to allow the relinking of a changed libopencm3.  This can be done by (for instance) providing a .o of your "private" code, and a makefile stub that relinks with a "new" libopencm3.  That's all that's required.

closed 2021-08-10 by karlp

doragasu, 2021-08-10:
Thanks for the reply. I supposed that is what I was going to get, but I had to try :-P

Unfortunately, providing the .o files is not always possible. In addition to being a hassle, I suppose that builds using lto (that we all use nowadays) might be relatively easy to decompile from object files (I have not tested this though, I might be perfectly wrong).

But of course this is not my code and I respect the work of the developers and the license is their choice.

Again thanks for the lib and the quick response.

// doragasu

CoryCline, 2022-12-09:
Well, I can confirm that I will never use this in a project under these licensing requirements. Thank you for clarifying.