libopencm3/libopencm3 #1366: stm32f405 rtc_time_set_time not working

aero141, 2021-11-19:
Using libopencm3 with gcc, gdb and Spark Fun STM32F405 Thing Plus the calls to rtc_time_set_time and rtc_calendar_set_date are not working.

	// Date: Friday 2021/12/31
	rtc_calendar_set_date(21, 12, 31, 5);
	// Time: 23:59:55 (11:59:55 PM)
	rtc_time_set_time(23, 59, 55, true);
but if I set the registers myself the correct time is set
	// Date: Friday 2021/12/31
	RTC_DR = 0x0021B231;
	// Time: 23:59:55
	RTC_TR = 0x00235955;
Raw TR: 00235959 Raw RTC_DR: 0021b231 Date: 2021/12/31 5    Time: 23:59:59
Raw TR: 00000000 Raw RTC_DR: 0022c101 Date: 2022/01/01 6    Time: 00:00:00
I've looked at the listing file and it appears to compile corectly.

Any Ideas?

slftr141@gmail.com

karlp, 2021-11-19:
can you show what the register is after the call that should  be working?  step through it and see what breaks?  I can't see from your paste what the values are you expect in each case.

aero141, 2021-11-19:
Hi Karl,
When I call the libopencm3 function rtc_calendar_set_date(21, 11, 18, 4);
I expected the RTC_DR to be 0x00219118 this bit pattern is the bcd date
with the day of week set to 4.
For the time rtc_time_set_time(23, 59, 55, true);
I expected the RTC_TR to be 0x00235955 (the bcd value for the time

The as run result with the calls to libopencm3 result in the following
output (1Hz hex printf of RTC_TR and RTC_DR):
Raw TR: 00000257 Raw RTC_DR: 0022c101 Date: 2022/01/01 6    Time: 00:02:57
Raw TR: 00000258 Raw RTC_DR: 0022c119 Date: 2022/01/19 6    Time: 00:02:58
Raw TR: 00000259 Raw RTC_DR: 0022c119 Date: 2022/01/19 6    Time: 00:02:59
Raw TR: 00000300 Raw RTC_DR: 0022c119 Date: 2022/01/19 6    Time: 00:03:00

But if I set the registers directly without calling rtc_calendar_set_date
or rtc_time_set_time like this:
// Date: Thursday 2021/11/18
RTC_DR = 0x00219118;
// Time: 23:59:55
RTC_TR = 0x00235955;
Then I get the following result:
Raw TR: 00235955 Raw RTC_DR: 00219118 Date: 2021/11/18 4    Time: 23:59:55
Raw TR: 00235956 Raw RTC_DR: 00219118 Date: 2021/11/18 4    Time: 23:59:56
Raw TR: 00235957 Raw RTC_DR: 00219118 Date: 2021/11/18 4    Time: 23:59:57
Raw TR: 00235958 Raw RTC_DR: 00219118 Date: 2021/11/18 4    Time: 23:59:58
Raw TR: 00235959 Raw RTC_DR: 00219118 Date: 2021/11/18 4    Time: 23:59:59
Raw TR: 00000000 Raw RTC_DR: 0021b119 Date: 2021/11/19 5    Time: 00:00:00
Raw TR: 00000001 Raw RTC_DR: 0021b119 Date: 2021/11/19 5    Time: 00:00:01
Raw TR: 00000002 Raw RTC_DR: 0021b119 Date: 2021/11/19 5    Time: 00:00:02
Which is exactly the behavior I want and expect from libopencm3

Patrick

aero141, 2021-11-22:
Okay, I've managed to set the date/time with calls to opencm3.  I was missing the calls to rtc_enable_bypass_shadow_registers(); and rtc_disable_bypass_shadow_register();
     The process to set the time/date is thus:
 1. rcc_periph_clock_enable(RCC_PWR);
 2. pwr_disable_backup_domain_write_protect();
 3. rcc_osc_on(RCC_LSE);
 4. rcc_wait_for_osc_ready(RCC_LSE);
 5. rcc_set_rtc_clock_source(RCC_LSE);
 6. rcc_enable_rtc_clock();
 7. rcc_periph_clock_enable(RCC_BKPSRAM);
 8. rcc_periph_clock_enable(RCC_RTC);
 9. rtc_unlock();
10. rtc_set_init_flag();
11. rtc_wait_for_init_ready();
12. rtc_set_prescaler(255, 127);
13. rtc_set_am_format();  // 24 hour format
14. rtc_enable_bypass_shadow_register();
15. rtc_calendar_set_date(21, 11, 21, 7); // Sunday 2021/11/21
16. rtc_time_set_time(15, 28, 00, true);  // 15:28:00  24 hour clock
17. rtc_disable_bypass_shadow_register();
18. rtc_clear_init_flag();
19. rtc_lock();

schodet, 2021-11-29:
@aero141 can we close the issue or do you see something to improve in libopencm3?

aero141, 2021-12-01:
We can close this issue, I'm not sure yet if I want to recommend a
documentation change or just leave it as is.  This issue was mostly my
learning how the STM32F4 RTC is initialized on power up.

Patrick

karlp, 2021-12-02:
Ok, well, when you want to improve the docs, you can just file a new one then, explicitly about the docs, rather than this issue we have "rtc doesn't work" if that's ok.

closed 2021-12-02 by karlp