libopencm3/libopencm3 #1380: stm32f411 blackpill stm32f413zh nucleo cannot read pin state changes gpio_get

bm16ton, 2022-02-01:
Hello, Im not an experienced programmer everything I know I learned from looking in the linux kernel so fair warning :) Ive been playing around with a firmware and kernel driver for usb gpio, currently basic gpio set/get/clear work, but Im having terrible issues reading buttons/pin states. The builtin button on the blackpill works but non of the other pins i have tried will change state. Ive tried setting the various,  builtin pull-up/pulldown inits, tried all the different ways I could wire the switch, but they always stay in whatever state they where initialized in. the one button on the black pill that works starts high and gets pulled low and the nucleo builtin button is opposite starts low gets pulled high, not sure if that info relates at all. Im positive im spiffing on something obvious here but just cant make it read. Ill post links to the files but whatever is in the files is just one of hundreds or more attempts to get it to work, 
https://github.com/bm16ton/i2c-star-blackpill-gpio/blob/master/i2c-star-stm32f411ceu6-blackpill/src/i2c-stm32f413zh-usb/i2c-stm32f413zh-usb.c
https://github.com/bm16ton/i2c-star-blackpill-gpio/blob/master/i2c-star-stm32f411ceu6-blackpill/src/i2c-stm32f4-usb/i2c-stm32f4-usb.c

any help would be greatly appreciated. just some extra info, the pins/buttons work fine in hal/arduino/mbed, in output mode the pins swich states perfectly fine. Ive tried a diff libopencm3 based firmware or two with same no pin reading results, so it hasta be me doing something dumb. The blackpill will enumerate with any libopencm3 version so tried the newest etc, the nucleo needs patches I applid and forgot about ages ago but the core lib isnt that old. I cant think of any other info that could be usefull at the moment. Thank you in advance for any possible insight.

karlp, 2022-02-01:
I'm pretty sure it's because you're comparing with 1.  Read the documentation on the gpio_get() method here: http://libopencm3.org/docs/latest/stm32f4/html/group__gpio__file.html#ga5fee90e8e5af7de567890ffae5ed50c8
you should be just doing someting like

```
if (gpio_get(GPIOC, GPIO13)) {
    // pin is set
} else {
   // pin is not set.
}
```
checking against 0 should work for not being set, but checking with "1" will only work for GPIO0 on any port.

closed 2022-02-01 by karlp

bm16ton, 2022-02-02:
Again sorry it took a day or two to respond but this was not the solution

bm16ton, 2022-02-02:
ah it was that! combined with a dead resistor, I think the dead resistor
threw me for the loop for so long