libopencm3/libopencm3 #1381: stm32f401cc tiny 401 wont enumerate

bm16ton, 2022-02-01:
I have a tiny f4101cc which ive always used interchangeably with my blackpill the same firmwares unmodified have always worked on both, but in libopencm3 the 401 wont enumerate...It doesnt even try my various pc's act as if nothing was ever detected. Ive tried the various usb patch fixes, which did work for several other boards, also tried the various vbus disable and force periph mode bits IE,
 	OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS;
        OTG_FS_GUSBCFG |= OTG_GUSBCFG_FDMOD | OTG_GUSBCFG_TRDT_MASK;
        OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS | OTG_GCCFG_PWRDWN;
	OTG_FS_GCCFG &= ~(OTG_GCCFG_VBUSBSEN | OTG_GCCFG_VBUSASEN);
etc. Where it doesnt even try to enumerate I assume the usb peripheral eiter isnt getting turned on/activated or the actual hardware has an issue (like weak or no pullups on D) thats is known and fixed via software hacks on the other sdks/ides kinda like the bluepill hacks we use, but this is best guess, and a guess from a foolish man at that. Any help would be greatly appreciated.
I believe this to be the exact model I have
https://stm32-base.org/boards/STM32F401CCU6-STM32-Mini-F401.html
nothing connected to the exra usb pins vbus etc. according to that link.

manuelbl, 2022-02-01:
I have a very similar board. According to the schematic they have in common that the CC1 and CC2 are connected and share a single 5.1kΩ resistor, which is wrong. If I connect it with a USB-C to USB-C cable to the USB-C port of a Mac or PC, the board does not get power. If I connect it through a USB hub with USB-A ports, it gets power.

Could the same be the cause with your board? As it has a power LED, it's easy to tell.

manuelbl, 2022-02-01:
Once the board has power, all that's needed is either PR #1256

Or:

```
#include <libopencm3/usb/dwc/otg_fs.h>

...

	usbd_dev = usbd_init( ...omitted... );

	OTG_FS_GCCFG |= OTG_GCCFG_VBDEN | OTG_GCCFG_PWRDWN;
```

karlp, 2022-02-02:
If there's no new issue here then, I think we can close this.

bm16ton, 2022-02-02:
All these solutions didnt work:( i know it took me a day or two to respond,
stupid life getting in way of projects. But the code still works on
blackpill 411 not greenpill 401, the usb cable amd port are fine. Ive tried
(and just retried) using exactly what was me tioned for usb bits amd still
dead. Whats the timeframe on having to respond before this gets closed?

karlp, 2022-02-03:
no problem, I just misunderstood :)

denis-shienkov, 2022-05-24:
Ahh, sorry, the trick with: `OTG_FS_GCCFG |= OTG_GCCFG_VBDEN | OTG_GCCFG_PWRDWN;` does work for me.

bm16ton, 2022-07-28:
Ok so after updating my libopencm3 and then applying these various patches (swear id tried'em several times before) It now works!! Def something different bout this board then even other ones that look identical, For instance sum with usb-c wont work on true usb-c ports but this one has the appropriate resistors (thought id mention the  little i knew). I just want to thank everyone who helped me and the developers. Im knew to all this, and oding in general and so far this is the only place anyone has given any support. Not sure if you know how important and how much it means to me and people like me. Again thank you!

closed 2022-09-01 by bm16ton