libopencm3/libopencm3 #1384: 'make html' doesn't generate Peripheral APIs

tomldavis, 2022-02-11:
When I build html docs, no html code is generated for the Peripheral APIs. Output from `make html` in the doc subdirectory includes multiple lines with:
`sh: dot: command not found`
as well as multiple warnings such as:
```
warning: Tag 'TCL_SUBST' at line 214 of file './Doxyfile_common' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"

```
`Doxygen version used: 1.9.3`
[doxygen.log](https://github.com/libopencm3/libopencm3/files/8050751/doxygen.log)
[image]

karlp, 2022-02-24:
try installing dot?  You can argue that you should get docs without diagrams, if you don't have dot, but honestly, I don't see why we need to support that level of customization.  The generated docs work fine locally and on www.libopencm3.org so I presume something is simply exiting early when it can't find dot.

tomldavis, 2022-02-24:
Karl,

Ok, I installed graphviz to get dot. I guess I’m too naive or I missed it in a Readme file, but I had no idea it was required to install graphviz just to get dot.

I’m still not getting the peripheral API’s. Now it looks like make can’t find the source files because the path it’s using is wrong. As far as I can see, there isn’t a permissions problem. I even built with sudo with no better results: 

	***@***.*** doc % sudo make TARGETS='stm32/f1 stm32/f3 stm32/f4' html

BTW, my Mac environment is pretty clean: I did a clean install (no migration). MacBook Pro (16-inch, 2021) running macOS 12.1.

Thanks,
Tom

karlp, 2022-02-25:
Have you actually run "make" on the project itself first? you need to build the library itself, so the docs can see which files are included where.

karlp, 2022-02-25:
Can you just do a full build instead of these attempts to use the partial builds? That's my last thought, otherwise, I can't reproduce it, so I'm not working on it.

karlp, 2022-02-27:
Of course not, but if it's an m1 mac issue, it's a fundamental m1 mac issue with doxygen or other tools, not anything we have any possible way of dealing with here.  The warnings you are getting have "input source /blah.c'" doens't exist, which looks like an absolute path, so I suspect something is very busted for you, well beyond the "make html" process.    If you clone to a new dir, do "make" in the top level dirtectory, then "make doc" in the top level directory, and it still isn't working, then you can share logs of that, but I'm not very convinced that there's anything wrong here yet.

tomldavis, 2022-02-28:
Karl,

Here’s what I did: 

***@***.*** Development % cd ~
***@***.*** ~ % git clone git://github.com/libopencm3/libopencm3.git
***@***.*** ~ % cd libopencm3
***@***.*** libopencm3 % make
***@***.*** libopencm3 % make doc

Still doesn’t work for ‘Peripheral APIs'. Looks like everything else is good (as before), e.g. 'Data Structures' and ‘Files' look good including dot output. 

I hope doing this in my home (~) directory is okay. Did you want this done some other directory?

BTW, building with libopencm3 and all other operations work great on my laptop. Running zsh with a really clean .zprofile.

***@***.*** libopencm3 % make -v
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
***@***.*** libopencm3 % doxygen -v
1.9.3

***@***.*** libopencm3 % dot -V
dot - graphviz version 2.50.0 (20211204.2007)

Repeated the process on my ancient Intel MacBook running macOS 10.15.7. Took a lot longer but got the identical result as far as I can tell.

Thanks,
Tom

tomldavis, 2022-02-28:
> I'll need more logs than
>
> if I'm meant to look at this :) You can use V=1 to get the commands run, and in every directory in doc you should get doxygen logs. Have look in them yourself, there might be something obvious.
>
> I suspect there might be something wrong with scripts/gendoxylist perhaps. You should be getting "doxy.sourcelist" files generated that have meaningful contents, like ~100plus lines per file,
>
>    189 ./stm32f0/doxy.sourcelist
>    178 ./stm32f1/doxy.sourcelist
>    181 ./stm32f2/doxy.sourcelist
>    181 ./stm32f3/doxy.sourcelist
>    236 ./stm32f4/doxy.sourcelist
>    214 ./stm32f7/doxy.sourcelist
>    152 ./stm32g0/doxy.sourcelist
>    160 ./stm32g4/doxy.sourcelist
>    133 ./stm32h7/doxy.sourcelist
>    182 ./stm32l0/doxy.sourcelist
>    177 ./stm32l1/doxy.sourcelist
>    198 ./stm32l4/doxy.sourcelist
>
> If those files are blank or substantially smaller, then you can see what's wrong with the script by hand:
>
> libopencm3/doc $ mkdir blah
> libopencm3/doc $ ../scripts/gendoxylist ../lib/stm32/f3 blah
> libopencm3/doc $ cat blah/doxy.sourcelist
> That script is meant to be posix sh, but there might be something that isn't working on zsh

Karl,

I’ll try to follow your instructions later today. Here’s what I get so far:

***@***.*** doc % cd /Users/tom/libopencm3/lib/stm32 
***@***.*** doc % wc -l ./stm32*/doxy.sourcelist
     189 ./stm32f0/doxy.sourcelist
     178 ./stm32f1/doxy.sourcelist
     181 ./stm32f2/doxy.sourcelist
     181 ./stm32f3/doxy.sourcelist
     236 ./stm32f4/doxy.sourcelist
     214 ./stm32f7/doxy.sourcelist
     152 ./stm32g0/doxy.sourcelist
     160 ./stm32g4/doxy.sourcelist
     133 ./stm32h7/doxy.sourcelist
     182 ./stm32l0/doxy.sourcelist
     177 ./stm32l1/doxy.sourcelist
     198 ./stm32l4/doxy.sourcelist
    2181 total

I’ve attached the stm32f1/doxy.sourcelist file. Looks completely reasonable to me. Seems the problem occurs downstream from gendoxylist.
 
FYI, that old Intel Mac is running bash so I don’t think it’s just a zsh issue. 

Tom

karlp, 2023-01-12:
ok, so.. there's doxygen logs in each directory?  what about them?  doc/stm32*/doxygen_*.log ?  try and look at one of them?  share one if you can't see anything obvious?

tomldavis, 2023-01-16:
Not sure why you’re sending this almost year after I last replied. I’m not actively using libopencm3 at this time. I just don’t have it in me to help anymore. Sorry.

Highly recommend you beg, borrow, steal or buy (!!!) a Mac so you can see what’s going on. You need to install Xcode (it’s free) and its command line tools to build libopencm3. Then all you need is the built-in terminal app and maybe a package manager like Homebrew to install your favorite Linux utilities. 

If you find a fix, please let me know and I’ll try it out.

Good luck,
Tom

closed 2023-01-16 by karlp