libopencm3/libopencm3 #1387: timer_clear_flag bug?

alwinl, 2022-02-20:
Inside the function timer_clear_flag (lib/stm32/common/timer_common_all.c:206 the status register is set to the negated passed parameter. Should this not be and-ed with the register content:
ie :
<code>
	TIM_SR(timer_peripheral) &= ~flag;
</code>
instead of
<code>
	TIM_SR(timer_peripheral) = ~flag;
</code>

aklomp, 2022-05-17:
From a quick skim of the `stm32f207` reference manual, the bits in the `TIMx_SR` register are marked as `rc_w0`. That means that the register can be read out as usual, but writes involve special semantics. Writing a bit to `0` lowers the bit, but writing a bit to `1` will not affect the existing value.

So the code is actually correct. It's writing the negated flag value to the register to clear those bits, while leaving the rest of the register intact.

karlp, 2022-05-17:
I'm going to close this, @aklomp  is correct, but for you to have filed this, I have a feeling you were running into problems?  (they probably weren't from _this_ particularly, but still)

If you were having real problems, please re-open, or make a new ticket with a clean story, your preference.    Thanks for looking at this @aklomp it slipped though my emails.

closed 2022-05-17 by karlp