libopencm3/libopencm3 #1390: Can't get ADC on STM32G4 to work

qosch, 2022-02-28:
Not sure this is the right place to ask but I don't know a better one:
I can't get the ADC to work. I started copying the example for STM32F4 series but when calling `adc_power_on(ADC1);` it keeps hanging in the line `while (!adc_is_power_on(adc));
`After consulting the reference manual I added various lines but the problem persists. Here is my ADC init function:
```
void adc_setup(void) {
    rcc_periph_clock_enable(RCC_GPIOA);
    gpio_mode_setup(CurrentFeedback1Port, GPIO_MODE_ANALOG, GPIO_PUPD_NONE, CurrentFeedback1Pin);

    rcc_periph_clock_enable(RCC_ADC1);
    rcc_periph_reset_pulse(RCC_ADC1);
    
    adc_disable_deeppwd(ADC1);
    adc_enable_regulator(ADC1);
    adc_enable_vrefint();

    //adc_disable_scan_mode(ADC1); //it seems this function does not exist for G4 series
    adc_set_sample_time_on_all_channels(ADC1, ADC_SMPR_SMP_6DOT5CYC);

    adc_power_on(ADC1);
}
```

I'd be willing to provide an example to the example repository if I get this working.

qosch, 2022-03-03:
I found the solution on stackoverflow: https://stackoverflow.com/questions/54856078/stm32l476-adc-not-ready
It is neccessary to choose an RCC peripheral clock using the ADC12SEL bits in the RCC_CCIPR register. Here is a complete ADC initialization and blocking read function (from the STM32F429 example) that works for me:

```
#include <libopencm3/stm32/rcc.h>
#include <libopencm3/stm32/adc.h>
#include <pins.h>

void rcc_set_adc12_clock_source(uint32_t clksel)
{
	RCC_CCIPR &= ~(RCC_CCIPR_SEL_MASK << RCC_CCIPR_ADC12_SHIFT);
	RCC_CCIPR |= (clksel << RCC_CCIPR_ADC12_SHIFT);
}

void adc_setup(void) {
    rcc_periph_clock_enable(RCC_GPIOA);
    gpio_mode_setup(CurrentFeedback1Port, GPIO_MODE_ANALOG, GPIO_PUPD_NONE, CurrentFeedback1Pin);

	rcc_set_adc12_clock_source(RCC_CCIPR_ADC12_PLLP);

	rcc_periph_clock_enable(RCC_ADC12);
	
	adc_disable_deeppwd(ADC1);
	adc_enable_regulator(ADC1);

    adc_power_on(ADC1);
}

uint16_t read_adc1_naive(uint8_t channel)
{
	uint8_t channel_array[16];
	channel_array[0] = channel;
	adc_set_regular_sequence(ADC1, 1, channel_array);
	adc_start_conversion_regular(ADC1);
	while (!adc_eoc(ADC1));
	uint16_t AdcResult = adc_read_regular(ADC1);
	return AdcResult;
}
```

I'll propose a pull request with the `rcc_set_adc12_clock_source` function added to include/libopencm3/stm32/g4/rcc.(c/h) if you agree. This problem should exist for most other peripherals as well (USART1-3, UART4-5, LPUART1, I2C1-3, LPTIM1, SAI1, I2S23, FDCAN, ADC34).

karlp, 2022-03-04:
not "should", rather "must" and therefore probably not as a sea of single use functions.  There's some proposals in the f7 or h7 PRs, and also for some L4 stuff, they need to be cleaned up and unified, but I'm not currently convinced that the sea of single use functions is a viable option.

karlp, 2022-03-04:
See https://github.com/libopencm3/libopencm3/pull/1315 for instance.

closed 2022-03-04 by karlp

qosch, 2022-03-05:
Alright, I'll wait for #1315 then.
I noticed that there are also many registers of the STM32G4 ADC missing (JQOVF, AWD3, AWD2, JEOS and JEOC from ADC_ISR register to only name the ones from the first register in the reference manual). I'd provide a pull request if you tell where you want them to go (a new adc_common_v3.h?).