libopencm3/libopencm3 #1392: Working stm32H7 spi ?

roman65536, 2022-03-05:
Hi, 

I wonder whether someone has an working spi on stm32h7 since libopencm3 does not support spiv3 yet ?
Code example would be great. 

thx & rgds
Roman

Inviz, 2022-03-08:
https://github.com/libopencm3/libopencm3/pull/1339

roman65536, 2022-03-08:
@Inviz I saw that as well, however, in master are the changes not yet integrated. Even with #1339 I had to fiddling around to make it work.

Inviz, 2022-03-09:
Yeah, that's unfortunate. I was just searching for stm32h7 support and found this issue and that issue, thought to post. Seems like painless h7 support is off the table for now. 

Did you make it work though? My h7 is gathering dust in the drawer. A little hard to tell what works on H7 and what doesnt.

roman65536, 2022-03-09:
Actually, In the meantime I could make it work. Part of the initialization I stole from [here](https://www.mikrocontroller.net/attachment/highlight/545366) now it works fine. If you wish can send you the stuff in two days or so.

slorquet, 2023-01-05:
Hi,

I am working on a professional H7 project and I wanted to use libopencm3 for it, so I need SPI to work, internal flash writes too.

Anyone else working on this? Because I will be, since it's a requirement for my project. We could work together.

"It starts to work" what works and what is missing at this point?

Thanks

roman65536, 2023-01-06:
@slorquet 
Here is what I did. Basically just init the spi and be able to send data @8bit width (to lcd display in my case), /CS is done in software.  Not sure anymore, why the flush was needed (may due to fullduplex and after write, I had to read from the spi data register). 

so here it goes:

void my_spi_flush(unsigned long spi)
{
  while((SPI_SR(spi) & SPI_SR_RXWNE) || ( (SPI_SR(spi) & (SPI_SR_RXPLVL_MASK<<SPI_SR_RXPLVL_SHIFT)) !=0 ) )
    SPI_RXDR(spi);
}

void my_spi_send8(unsigned long spi,unsigned char d)
{
  SPI_CR1(SPI2) |= SPI_CR1_CSTART;
  while(!(SPI_SR(spi) & SPI_SR_TXP));
  SPI_TXDR8(spi)=d;
  while( !( SPI_SR(spi) & SPI_SR_TXC));
  my_spi_flush(spi);
  //spi_wait_for_transfer_finished(spi);
  //SPI_IFCR(spi)=SPI_IFCR_EOTC;
}

void spi_setup()
{

  //rcc_set_spi123_clksel(RCC_D2CCIP1R_SPI123SEL_PLL1Q);
	rcc_periph_clock_enable(RCC_SPI2);
	rcc_periph_reset_pulse(RST_SPI2);
	
    gpio_mode_setup(GPIOB, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO12|GPIO13|GPIO14|GPIO15 );
		 gpio_set_output_options(GPIOB, GPIO_OTYPE_PP, GPIO_OSPEED_100MHZ,GPIO12|GPIO13|GPIO14|GPIO15);
		//		gpio_mode_setup(GPIOA, GPIO_MODE_INPUT, GPIO_PUPD_NONE , GPIO6); 
        gpio_set_af(GPIOB, GPIO_AF5, GPIO12|GPIO13|GPIO14|GPIO15);

		gpio_mode_setup(GPIOB, GPIO_MODE_OUTPUT, GPIO_PUPD_NONE,GPIO12|GPIO8|GPIO9);
		gpio_set_output_options(GPIOB, GPIO_OTYPE_PP, GPIO_OSPEED_100MHZ,GPIO12|GPIO8|GPIO9);

		
		//		 gpio_mode_setup(GPIOA, GPIO_MODE_AF, GPIO_PUPD_PULLUP, GPIO15 );
		// gpio_set_output_options(GPIOA, GPIO_OTYPE_PP, GPIO_OSPEED_100MHZ, GPIO15);
		//		gpio_mode_setup(GPIOA, GPIO_MODE_INPUT, GPIO_PUPD_NONE , GPIO6); 
        //gpio_set_af(GPIOA, GPIO_AF5, GPIO15);

		spi_reset(SPI2);
		SPI_IFCR(SPI2)|=SPI_IER_MODFIE;

		//spi_init_master(SPI1,5,0,0,0);
		spi_disable_crc(SPI2);
		spi_enable_software_slave_management(SPI2);
		//spi_set_transfer_size(SPI2, 8);
		spi_disable_ss_output(SPI2);
		spi_set_full_duplex_mode(SPI2);
		spi_set_transfer_size(SPI2, 0x0);
		spi_send_msb_first(SPI2);
		//spi_set_bidirectional_transmit_only_mode(SPI2);
   		spi_set_nss_high(SPI2);
        //	spi_init_master(SPI1,SPI_CFG1_MBR_CLK_DIV_256,0,0,0);
		spi_set_baudrate_prescaler(SPI2, SPI_CFG1_MBR_CLK_DIV_2);
		spi_set_data_size(SPI2,SPI_CFG1_DSIZE_8BIT);
			spi_set_master_mode(SPI2);
			//			SPI_IFCR(SPI1)|=SPI_IER_MODFIE;
			SPI_CR2(SPI2) =0;
			SPI_CFG2(SPI2) |= SPI_CFG2_AFCNTR | SPI_CFG2_SSOE ; 
		spi_enable(SPI2);
		SPI_CR1(SPI2) |= SPI_CR1_CSTART;
		my_spi_send8(SPI2, 0x55);
		my_spi_send8(SPI2, 0xaa);
		my_spi_send8(SPI2, 0x01);
		my_spi_send8(SPI2, 0x10);
		my_spi_send8(SPI2, 0x81);
		my_spi_send8(SPI2, 0x99);
		
		//	spi_wait_for_transfer_finished(SPI2);
		
}

rgds
Roman

slorquet, 2023-01-10:
unfortunately raw code like this does not really help since it does not come with the required libopencm3 infrastructure. I'll have a look a bit later but it would be better to have a well formatted pull request we can test.

As I understood stm32h7 has a new SPI IP so it needs the specific functions in a new file while keeping the common code separate, etc.