libopencm3/libopencm3 #1400: Compile failure under gcc-arm-11.2-2022.02-x86_64-arm-none-eabi

rgwynnjones, 2022-04-05:
I'm new here so this could all be wrong.
Building indirectly from stm32-vcu, using a new dev setup with latest gcc cross compiler for ARM from here;
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads
which at time of writing is 11.2-2022.02
When compile get the following error;

<pre><font color="#4E9A06"><b>rjones@sparrow</b></font>:<font color="#3465A4"><b>~/git/Stm32-vcu</b></font>$ make V=1
  CPP     obj/stm32_can.o
arm-none-eabi-g++ -Os -Wall -Wextra -Werror -Ilibopeninv/include -Iinclude/ -Ilibopencm3/include -fno-common -std=c++17 -pedantic -DSTM32F1 -DT_DEBUG=0 -ffunction-sections -fdata-sections -fno-builtin -fno-rtti -fno-exceptions -fno-unwind-tables -mcpu=cortex-m3 -mthumb -ggdb3 -o obj/stm32_can.o -c libopeninv/src/stm32_can.cpp
<b>libopeninv/src/stm32_can.cpp:</b> In member function &apos;<b>void Can::Save()</b>&apos;:
<b>libopeninv/src/stm32_can.cpp:47:51:</b> <font color="#CC0000"><b>error: </b></font>expression does not compute the number of elements in this array; element type is &apos;<b>Can::CANIDMAP</b>&apos;, not &apos;<b>uint32_t</b>&apos; {aka &apos;<b>long unsigned int</b>&apos;} [<font color="#CC0000"><b>-Werror=sizeof-array-div</b></font>]
   47 | #define SENDMAP_WORDS         (<font color="#CC0000"><b>sizeof(canSendMap) / sizeof(uint32_t)</b></font>)
      |                                <font color="#CC0000"><b>~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~</b></font>
<b>libopeninv/src/stm32_can.cpp:204:57:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro &apos;<b>SENDMAP_WORDS</b>&apos;
  204 |    SaveToFlash(SENDMAP_ADDRESS, (uint32_t *)canSendMap, <font color="#06989A"><b>SENDMAP_WORDS</b></font>);
      |                                                         <font color="#06989A"><b>^~~~~~~~~~~~~</b></font>
</pre>

And more similar errors and warnings in the same source file.

More details
Fresh ubuntu 20.04.4 install, stm32-vcu git checkout which collects libopencm3 as a dependency.

<pre><font color="#4E9A06"><b>rjones@sparrow</b></font>:<font color="#3465A4"><b>~/git/Stm32-vcu</b></font>$ which arm-none-eabi-gcc
/opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
<font color="#4E9A06"><b>rjones@sparrow</b></font>:<font color="#3465A4"><b>~/git/Stm32-vcu</b></font>$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 11.2.1 20220111</pre>

rgwynnjones, 2022-04-05:
As the compiler suggests later, added parens to sizeof(uint32_t) resolves the compilation issue.
e.g working new code at stm32_can.cpp:47

#define SENDMAP_WORDS         (sizeof(canSendMap) / (sizeof(uint32_t)))
#define RECVMAP_WORDS         (sizeof(canRecvMap) / (sizeof(uint32_t)))

karlp, 2022-04-05:
You might be using libopencm3, but the linked code is not part of the libopencm3 project, it looks like some project called "libopeninv" ?  Perhaps try filing the ticket there.

closed 2022-04-05 by karlp