libopencm3/libopencm3 #1406: Atmel SAMD21 support

doxygen-spammer, 2022-04-26:
Hi! I have some evaluation boards with SAMD21 and SAMR21, which I would like to program with a free framework (i. e. libopencm3).

I understand that support for SAM is still incomplete, and for SAMD21 in particular there is only PORT.

There is currently a directory for SAMD, containing the interrupt table for SAMD09/10/11, and the library for PORT.
The interrupt tables of SAMD20 and SAMD21 are incompatible, each, so they would need either `#if` stuff or separate files.

I would like to implement more support for SAMD21.
This is what I would like to do first:

* Interrupt vector table (need to figure out where irq.json is processed)
* Library for PM (Power Manager, turns on/off peripherals like TC, SERCOM), can be ported from SAM4L, probably by making most code “common”.
* Library for EIC (interrupts from pins)
* Library for TC (timer), some register definitions from SAM3X available, do not apply here.

If I have still fun after that, I plan these:

* SERCOM: some library for USART and some register definitions for UART available, but they do not apply here. Apparently they were relevant for SAM3X once.
* ADC: some library for ADCIFE available for SAM4L, will figure out if that is the same.

Please let me know what you think about this.

I have read HACKING_COMMON_DOC.
Are there other guidelines or quality standards for adding support for new families, which I should read?

karlp, 2022-04-27:
knock yourself out!

Use separate trees where required, rather than ifdefs generally, especially when you're already talking about different families.  Just use common dirs where possible to share peripheral support code, as they are normally shared across families within the same vendor.  only thing that people sometimes miss is running "make doc" and "make stylecheck" to check that the doxygen is building nicely.

closed 2022-04-27 by karlp

doxygen-spammer, 2022-05-08:
I decided to make a new family in the directory sam/d21.
SAMD is a similar family like STM32F, where e. g. SAMD10 and SAMD21 or STM32F0 
and STM32F1 are to wide extents compatible, but differ in details like memory 
addresses.
So I adapt the logic from STM32 to SAM.

The directory sam/d can stay for now, because D09, D10, and D11 are equivalent 
as far as already covered (interrupt vectors and PORT).

---

But I am not sure about the doxygen groups, which would be affected by a new 
family.

If I go to doc/samd/html/modules.html, I can see the groups “Peripheral APIs” 
and “SAMD Defines”, which are defined in doc-samd.h as `peripheral_apis` and 
`SAMD_defines`.
Currently there is the library for PORT, which appears as “IO Port API” and 
“IO Port Definitions” in these two groups.

But for SAM3x and SAM4x families, there is not even a “Peripheral APIs” group, 
although GPIO libraries are implemented.
Documentation for these libraries is therefore not accessible.

In the STM32 families it looks better.
Files like f1/gpio.h define a group “GPIO Defines” as subgroup of “STM32F1xx 
Defines”, and then include common/gpio_common_all.h, which uses the group 
“GPIO Defines”.

I guess the SAM families should have operational doxygen modules like the 
STM32 families, so I am going to modify the doxygen groups of sam/*.