libopencm3/libopencm3 #1415: stm32f4: USB isochronous stream is not implemented for USB OTG {FS|HS} driver

denis-shienkov, 2022-05-27:
Right now the library does not handles the isochronous endpoints properly in the DWC USB driver (it belongs and tested at least on the stm32f407 && stm32f401 mcu's).

I have enabled support for the input endpoints (my personal project requires only the IN eps at this time), like this: https://github.com/denis-shienkov/libopencm3/commit/5c8bc982e5a94e3c440d26f5f762db9c5f13c040

EV-Builder, 2023-01-12:
did you solve this issue?

karlp, 2023-01-12:
Oh, further, as this is just an issue pointing out things that are unimplemented, not _bugs_ per se, I'm closing this, we can't have an issue for every single missing feature we'd drown.

closed 2023-01-12 by karlp

denis-shienkov, 2023-01-12:
LOL, how you can close the issue which is not fixed? 

> I would need isochronous test code

What is a problem to do it himself? Because I have not an ideas how to do these tests.

UPD: Right now, this `libopencm3` library is unusable at all. I don't understand, why the ISO stream did not implemented  whithin a long time ago... That's wery fun. 

Anyway, right now I use a HAL instead, where all things are implemented.

karlp, 2023-01-12:
Should we have an issue for every single feature that is unimplemented?  For how long?  How can that possibly be useful?  Would one issue that just says "library is not complete" be ok?  The reason I close "feature not impelemnted" tickets is because it starts providing a view that "only features with issues" are unimplemented, which isn't true at all, it's just "some missing features were reported as missing"

As for "do it himself" tests, that's just it, a maintainer needs isochronous usb test code to test, evaluate and maintain isochronous usb support code going forward.  Without that, it won't get merged.