libopencm3/libopencm3 #1427: ABP1 Peripherals not working

shane-woods, 2022-07-13:
Hello, I've been working with this lib for a while and been using it to program a STM32F0. We've used all these peripherals successfully:

SPI1
USART1
TIM1
TIM15

We have tried to setup SPI2 for some time now and haven't been able to get it to work. We have set it up the same way as SPI1. We thought this might be because SPI2_RCC is on ABP1 and all the other peripherals we have working are on ABP2. Is there any function we should be calling to enable clocks for ABP1 peripherals? Anything helps

sbrk-r, 2022-07-15:
It might sound stupid, but I remember running into a similar problem, only to finally realize none of my STM32F0 even *have* SPI2. The datasheet is a bit misleading ("up to two SPI"), so double check that.

For STM32F030 and 070 anyway, only the ones with the biggest flash have SPI2... not sure how that is logical, but that's what ST does.

If this is the case you can use USART2 in synchronous mode (though the maximum speed is quite a bit less, 2 or 3MHz?).

karlp, 2022-07-15:
If the above suggestions don'thelp, you'llneed to share some code samples I think.

shane-woods, 2022-07-15:
Interesting. I'll have to look at the rm and see what the case is. If I️ can't find anything I'll come back with code.

Rajshekhar1208, 2025-01-21:
The STM32 F0C8/CC/R8/RC do have SPI2

karlp, 2025-01-21:
no code, no follow up. closing this.

closed 2025-01-21 by karlp