libopencm3/libopencm3 #1434: Differences required in usb callbacks for stm32f1 and stm32f4?

bm16ton, 2022-09-01:
Hello,
       I have a homebrew firmware and kernel driver for a usb to adc. It works perfect on my f411/f401 in a composite device setup but the f1 refuses to enumerate. more info
endpoint 1 is i2c-star with 1 interrupt endpoint (unused everything is done endpoint 0)
endpoint 2 is usb to gpio/irq 1 interrupt (interrupt endpoint only sends for irq to fire in kernel driver)
endpoint 3 is the usb to adc 2 bulk 1in 1out,

on linux F1 fails to enumerate "can't set config #1, error -110" 
everything during setup is fine until its sent the brequest to switch to config 1 then the device stops responding. Using redneck methods i narrowed he issue down to a single line in set_config

void usbadc_set_config(usbd_device *dev, uint16_t wValue)
{
//	(void)wValue;
    configured = wValue;

		usbd_ep_setup(dev, 0x03, USB_ENDPOINT_ATTR_BULK, BULK_EP_MAXPACKET,
			gadget0_ss_out_cb);
		usbd_ep_setup(dev, 0x83, USB_ENDPOINT_ATTR_BULK, BULK_EP_MAXPACKET,
			gadget0_ss_in_cb);
		usbd_register_control_callback(
			dev,
			USB_REQ_TYPE_VENDOR | USB_REQ_TYPE_INTERFACE,
			USB_REQ_TYPE_TYPE | USB_REQ_TYPE_RECIPIENT,
			gadget0_control_request);
		/* Prime source for IN data. */
		gadget0_ss_in_cb(usbdev, 0x83);
}

I shamelessly copied the bulk endpoint stuff from gadget 0  hence the names. The very last line " gadget0_ss_in_cb(usbdev, 0x83); " is what keeps it from enumerating if i remove it it enumerates fine but the usb adc doesnt work, the F4 works fine with that line in place but also fails to work with that line removed. Both f4 and f1 use same code/files except some pre-processors around gpio setup (which ive removed for testing no affect) and the usb driver used. So i assume Im clueless about something obvious and well known about the differences between the 2 usb drivers and/or usb hardware and just being dumb. Any help would be super appreciated. I can supply sum (even all) code if needed.

bm16ton, 2022-09-04:
me being dumb had the f103 using a adc_set_regular_sequence cp'd from the f4 code, removed that all is well. Feel so dumb starting this issue now.

closed 2022-09-04 by bm16ton

karlp, 2022-09-04:
great :) because I didn't really have any great ideas, was going to have a better look when I was home again!

bm16ton, 2022-09-04:
Its amazing to me that someone as busy and talented as yourself would
waiste time on someone like me, truly are a good man my friend and again
thank you for all your hard work. Switching from combining arduino examples
to using libopencm3 these past few months have made this aging dogs life so
much more fun and interesting!