libopencm3/libopencm3 #1437: Generate linker script for backup SRAM

ekawahyu, 2022-09-13:
I use STM32F4-Discovery and tweak the linker script to generate backup SRAM section (by piggy-back it over stm32f4ccm family. I am wondering if there is a work-in-progress about this, or is there any reason why backup SRAM isn't part of libopencm3 linker script generator? Thank you.

karlp, 2022-09-13:
Can you share your tweak so we better understand your issue?

karlp, 2022-09-13:
It's probably because it only appears to be on _some_ F4 parts, F2 parts, and some F7 parts, and no-one's needed it for anything.  It's probably not a hugely complex task, you would just need to add more fields to the linker devices.data file, and another section.  You can probably just use one of the SRAM2.3.4.5 fields that are currently unused for your parts.    See https://github.com/libopencm3/libopencm3/blob/master/ld/devices.data#L154 and https://github.com/libopencm3/libopencm3/blob/master/ld/devices.data#L588

You'll still have to place variables into that section yourself though.  I'm closing this now as it's a new/incomplete feature wish, and we can't keep tracking items for every one of those...

closed 2022-09-13 by karlp

ekawahyu, 2022-09-13:
Yeah, probably it is not a good fit, since the backup SRAM requires initialization to disable write protection after reset. I just thought that someone may have some work-in-progress on it and would like to see if it is the plan. Thank you for responding Karl.

karlp, 2022-09-13:
Eh, it's pretty harmless to just have zones in the linker script, I'm not at all opposed to adding them, and it would be certainly valid to add code into, maybe rcc.c or flash.c or pwr.c to provider a helper to turn it on/off sort of thing, it would definitely not be "in the way" of anything.

ekawahyu, 2022-09-13:
Hmmm, actually it is kind of hard to pull information on ST website of all STM32s that have backup SRAM. I am trying to make this generic on a branch, so that any STM32 variant with backup SRAM can benefit from it. It really is hard to navigate the product selector from ST website. Do you know a better (or faster) way to do it?