libopencm3/libopencm3 #1453: Avoid usage of initialized data ?

slorquet, 2023-01-10:
Hi,

I want to use libopencm3 for a bootloader project.
After the bootloader is built it will be merged with the main code (nuttx rtos based) using some objcopy magic.

This method would work best if I could avoid initialized data in libopencm3, which is the only place that define contents in the .data section.

I'm using a stm32, and the only libopencm3 code that uses initialized data is rcc.c that stores the clock tree (eg, list of clock rates).
This clock tree is initialized in the file via a data initializer, which uses the data section.

This could be avoided if the initializer was replaced by a function (eg rcc_setup_default, or rcc_init) that populates the clock tree default value, instead of initialized data.

I see that pre_main defined here could be the place to automatically call rcc_init.
https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/h7/vector_chipset.c

What do you think about this? Could this be done inconditionnally?

I'm willing to do the changes for all stm32 platforms. Would you accept such a change?

Thanks for your attention.

slorquet, 2023-01-10:
Hi, I am talking of:
https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/h7/rcc.c#L35

This is the only place in the full libopencm3 basic chip support routines that use initialized vars.

objcopy itself is not the main reason for a change here.

Writing a bootloader means I would like something as self contained as possible.

slorquet, 2023-01-10:
screw it. I need to use littlefs in this project, and even with the correct defines to avoid them, this thing uses assert and malloc, which in turn brings the newlib impure stuff into the link and I cant get around it, so it means I have to deal with initialized data after all. Thats a shame.

It's not going to be easy.

slorquet, 2023-01-10:
I will not be able to what I want. I managed to add the bootloader code to my application ELF file because the ELF file already has an empty memory reservation for where the bootloader resides.
However, massaging the ELF file to add the initialized data section does not work for several reasons
-there is no segment associated to this new section: This means that the loader (openocd or whatever) will not load this segment into memory when programming
-the --change-section-lma option of objcopy has no effect, it is not possible to add a section to an ELF file having different load and execution addresses.

So i'm back to square one: I cannot use newlib functions (malloc, etc) in my bootloader, which are required for littlefs. So, I can ditch littlefs entirely. I'll use a raw flash zone instead.

It also means that I am indeed interested in avoiding initialized data in rcc.c. I'm sending an pull request.

slorquet, 2023-01-11:
I got everything working by putting the whole binary file generated by libopencm3 objcopy (which contains all required loadable sections from the elf file) in the pre-allocated boot section of my final executable. This way I dont have to add complex new sections to my elf file and mess with section addresses.

Despite this, I think it's still a good thing to avoid having an initialized var just for RCC when nothing else in libopencm3 requires initialized data.

karlp, 2023-01-12:
I have no particular objections to doing heavy work on the h7 code, it came in from people who wanted to do things differently, and it hasn't worked out well.  I'm glad I wasn't involved in trying to put your bootloader into actual sections though, stuffing it in as a bin sounds farrrrrr simpler, (and what I've done in the past)

karlp, 2023-01-12:
soundsd like we can close this as well now, let me know if you want to re-open this.

slorquet, 2023-01-12:
Yes it is farrrr simpler now that I understood that I should embed the full binary image instead of just the .text section.

I have closed the pull request and will be closing this now,

Still interested by the other pending h7 changes.

closed 2023-01-12 by slorquet