libopencm3/libopencm3 #1457: Coordination for all stm32h7 porting efforts

slorquet, 2023-01-12:
Hi,

It seems that several people are interested in getting stm32h7 to work better, specifically spi, and i2c.

Several issues and pull requests have been opened for this purpose:
https://github.com/libopencm3/libopencm3/issues/1455 Arduino Portenta h7 additions
https://github.com/libopencm3/libopencm3/issues/1392 Working stm32H7 spi ?
https://github.com/libopencm3/libopencm3/pull/1452 Enable required files to ensure successful build of a "stm32h7" only lib
https://github.com/libopencm3/libopencm3/pull/1456 stm32: When setting baudrate of a USART, first get the clock frequency for that USART.

The important point is the code architecture: libopencm3 has files that offer functions common to all or a subrange of the stm32 chips, and files that are specific to a family.
I dont have enough expertise to determine how the split should be really done.

My questions are
-is the stm32h7 using a specific v3 spi block different from other platforms? or can we reuse spiv2
-what functions can be kept in the common files and what functions have to be put in a separate file?

The same questions are valid for i2c, which is an equally important communication block.

Since we are several people interested in this I assume some collaboration is possible.

calling @bm16ton @roman65536 and @ventZI, who seem to be the most advanced on this question.

bm16ton, 2023-01-12:
Oh wow thank you for responding to my previous issue/post its appreciated! The spi v3 is a beast, fifo based, which is fine but the errata makes it difficult. Having to disable spi after every transaction, set the t-size etc re-enable etc is where i think im going wrong. Setting the tsize to unlimited should work around that but i havent gotten that to work either. Its very possible i hhave sum stupidlysimple thing set wrong and im beating the wrong horses on it. Unfortunatly i dont think spi v3 shares much with the previous v2. I hhavent done much testing but i2c v2 seems to be working but alas ive always had issues with i2c v2 so well see once i gget more miles on it. Also on my list to tackle is dma. With 2 dma muxes sum work will be required but i do think (based of a very quick look) that the current code for dmamux (or at least parts) will be reusable so thats nice. When i started i thought usbhs was gonna be my biggest hurdle but that took 15min, but spi is my problem child facepalm. I did find sum potential issues and even tho its not working ill throw a logic analyzer on it hopefuly later today and see if its actually transmitting anything now. The qspi flash isnt working for either but i hhavent throw much effort at it yet. So yeah even tho im way underqualified to do so im gonna focus on spi and dma for now. Fingers crossed. Be nice if those of us working on the h7 had a central place to communicate and share, combine code without spamming the issues area lol. Again thank you for the response!

slorquet, 2023-01-12:
Issues are fine for discussing code :) Communication is more efficient when it is not dispatched in several places. (note: I wont go to any discord)

bm16ton, 2023-01-12:
funny you say that i finally broke down and decided to try it, went to
install it in playstore and all the artwork/design reminded me if a 5th
grade class room and i said no and stopped. I like gist but it seems all
libopencm3 conversations have stopped. Being a guy who combined a couple
arduino sketches then jumped to libopen has been tuff. Resources have been
slim. My current method for anything new/unsupported has been go bare metal
get it working.Then replace everything i can with whats already in libopen,
then add functions for whats left over. Big hurdles for on old redneck
learning new tricks, but very rewarding

roman65536, 2023-01-12:
@bm16ton 
Have a working qspi if you are interested. But is outside libopencm3 structure/tree (since I was interested trying out mapping and later on read/write/erase using ext2 fs) and once it worked, I didn't bother to put it to libopencm3.
However qspi very depends on the qspi device, resp. qspi device capabilities (instruction in 1bit/4bit mode, address length 16,24,32 bit, address mode 1,4 bit etc). 

rgds
Roman

bm16ton, 2023-01-12:
Omg! yes please i would very much like to see/use that code. Ive been
working on spi mostly. I had tested my spi device on tge pins im using and
it worked before my attempts in libopen 2days later i test it
again....dead. God only knows when it died or how many of my attempts may
have worked lol. I digress yes please that would be great. Glad were all
coming together on this

bm16ton, 2023-01-12:
needed to write to the rtc's backup registers for bootloader. All the registers i needed etc matched up. If i find time tonight ill check the rest its possible that it would be a very quick ez port. Sidenote my H7 is ulpi usbhs only, I need to check how many usb endpoints the usbfs the various h7s have. Ive found what looks like a static 4 in places representing the amount of endpoints. But my f413zh has 6. If the h7's have more as well itll make any fix i come up with go a diff way.

roman65536, 2023-01-13:
@bm16ton here we go : 
I use "W25Q64" device, so to see what is going on, you need the datasheet. I used this :https://www.pjrc.com/store/w25q64fv.pdf

In quad_map() is inline more explanation how and why to setup h7's qspi registers. 

@slorquet feel free to integrate it (somehow) to libopencm3 if you like. I know, I did it the quick and dirty way, 
this could be implemented much better and universaly where you could somehow define the flash instructions for enable qspi, read qspi, write qspi, erase etc. 

And sorry for just copy pasting it here.

rgds
Roman

void quad_setup(void)
{
/* the usual enable and setup pins for qspi
*/
	rcc_periph_clock_enable(RCC_QSPI);
	rcc_periph_reset_pulse(RST_QSPI);
	gpio_mode_setup(GPIOD, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO11| GPIO12| GPIO13 );
        gpio_set_output_options(GPIOD, GPIO_OTYPE_PP, GPIO_OSPEED_100MHZ, GPIO11| GPIO12| GPIO13);
        gpio_set_af(GPIOD, GPIO_AF9, GPIO11| GPIO12| GPIO13);
	gpio_mode_setup(GPIOB, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO2|GPIO6 );
        gpio_set_output_options(GPIOB, GPIO_OTYPE_PP, GPIO_OSPEED_100MHZ, GPIO2|GPIO6);
        gpio_set_af(GPIOB, GPIO_AF9, GPIO2);	
        gpio_set_af(GPIOB, GPIO_AF10, GPIO6);	
	gpio_mode_setup(GPIOE, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO2 );
        gpio_set_output_options(GPIOE, GPIO_OTYPE_PP, GPIO_OSPEED_100MHZ, GPIO2);
        gpio_set_af(GPIOE, GPIO_AF9, GPIO2);	

/* Not sure anymore why, but it works :)
    I guess clock prescale for spi mode (generaly flash starts in spi mode and we have to turn on the qspi mode on)
  address len needs to be set, so the mcu knows about the size of flash for mapping
*/

 	QUADSPI_CR= 80<<QUADSPI_CR_PRESCALE_SHIFT 
		    | 3 << QUADSPI_CR_FTHRES_SHIFT
		    |    QUADSPI_CR_EN ;
	QUADSPI_DCR = 22<<QUADSPI_DCR_FSIZE_SHIFT
			| 7 << QUADSPI_DCR_CSHT_SHIFT ;
	QUADSPI_ABR =0;
	QUADSPI_DLR =16;
	QUADSPI_AR =0;
	
	
	QUADSPI_CCR=0x00000166;             // enable reset 66  Page 70 in the datasheet
	while( (QUADSPI_SR & 32) == 32 );
	QUADSPI_CCR=0x00000199;            // soft reset  Page 70 
	while( (QUADSPI_SR & 32) == 32 );
	while(!(QUADSPI_SR & 0x04));
	while(!(QUADSPI_SR & 0x02)); /* Wait for TCF flag to be set */
	QUADSPI_CCR=0x00000138;            // enable quadspi      Page 68
	while( (QUADSPI_SR & 32) == 32 );
	while(!(QUADSPI_SR & 0x02)); /* Wait for TCF flag to be set */

/* W25Q64 is now in the qspi mode
*/
}

quad_map()
{
/* Page 31 in the data sheet
    following keep in mind : 
   we are sending command in 4 bit mode, we are sending address in 4 bit mode, we have 2 dummy cycle in 4 bit mode and we receive data in 4 bit mode. 
the Address len acording datasheet 24bit aka 3 bytes 
and we'd like to map the flash to memory.

datasheet of h7:
 FMODE : 3 Memory map
 DCYC:  2 dummy cycles
[... 159 lines cut ...]
			//	QUADSPI_ABR =0;
	QUADSPI_AR = addr;

		while( (QUADSPI_SR & 32) == 32 );

	    	QUADSPI_CR &= ~  QUADSPI_CR_EN;
			QUADSPI_FCR=2;          //CLEAR CTFC Flag
			
    	QUADSPI_CR |=   QUADSPI_CR_EN;
	QUADSPI_CCR = 0 << QUADSPI_CCR_FMODE_SHIFT
			| 0 << QUADSPI_CCR_DCYC_SHIFT
			| 0 << QUADSPI_CCR_ABMODE_SHIFT
			| 0 << QUADSPI_CCR_ABSIZE_SHIFT
			| 0 << QUADSPI_CCR_ADMODE_SHIFT
			| 2 << QUADSPI_CCR_ADSIZE_SHIFT
			| 3 << QUADSPI_CCR_IMODE_SHIFT
			| 0 << QUADSPI_CCR_DMODE_SHIFT
		//	| QUADSPI_CCR_SIOO
		//	| QUADSPI_CCR_DDRM
	  | 0x04 ;                    //  quad write disable in qspi 
			
	while( (QUADSPI_SR & 32) == 32 );
			QUADSPI_FCR=2;          //CLEAR CTFC Flag
    QUADSPI_CR &= ~  QUADSPI_CR_EN;
}

bm16ton, 2023-01-14:
Damn this is awesime thank you. Its ague but if memory serves 128mb can mapped to system (read only) then the rest is all indirect access after that. If thats correct thats more then fine by me. The portenta has a MX25L12833FZ2I. I already have the ez clock functions for qspi working.I know sum level of chip specific settings can be read/determined or maybe culled from a list at compile time with model passed. Im fairly certain the wear leveling stuff is complicated af so im problu a long ways off from tackling that. But im not sure the mmost common use cases so not sure how much a priority it should have.My gut says probly pretty high. I kinda feel were making sum progress, but this is the most collaboration ive ever been involved in so...

slorquet, 2023-01-18:
Hi,

Tomorrow I will need to work on stm32h7 spi, so can you tell me @ventZl if you intent to push your working SPI (not QSPI) code today, or if I just reimplement it tomorrow? Unfinished is fine.

Either solution is fine, no pressure, but I would just like to know so I can plan my day,

Thanks,
Sebastien

roman65536, 2023-01-18:
@slorquet be sure you have a logic analyzer or dso right next to you :) to double check. 
Code from #1392 should work, may be there is one little issue. Since my example is fullduplex, I check after every write whether there are data from receiver. This is probably not necessary when spi is set to tx only. 

Also not sure how do you like to handle spi packetwise communication, which would be great for lcd displays ? But maybe you already have some sort of concept how to put it all together.

rgds
Roman

slorquet, 2023-01-18:
I'll have a look, also I have a working driver for NuttX. I do not want to copy code, but I can still observe the proper sequencing of register accesses.

What I'm primarily concerned about is write a properly integrated spiv3 driver within libopencm3, so raw code gives a hand but is not what I need. I must separate which functions will go inside the common files and which ones are specific to the stm32h7.

I'll have a fx2lafw analyzer with sigrok pulseview.

slorquet, 2023-01-19:
Working now on the code structure for stm32h7 spi

Here is my plan

spi_common_all.c cannot be used for spiv3 since so many things are different, register names, bit definitions, transfer algorithms.

so I've made a new file spi_common_v12.c with most routines from spi_common_all, except reset which is the actually only common routine.

I have then started spi_common_v3 for the stm32h7, rewriting most of the contents in spi_common_v12 (aka _all)

The pull request sent by @ventZl , even if surprising, makes a lot of sense but has errors and is missing important bits to allow spi transfer.

The routines are not working yet but there is progress, both in code structure and actual spi behaviour.

For your information here are the main changes I've done today, I'm not sending a pull request yet because it's not ready
https://github.com/libopencm3/libopencm3/compare/master...slorquet:libopencm3:stm32h7-spi

@karlp I know you are very busy, and I would like to be allowed to push trivial changes like the breaking uart thing. Because obviously you dont have enough time for this project, which cannot be blamed and is very understandable, but is also blocking the progress of the project.

roman65536, 2023-01-19:
@slorquet went quickly over your code, is cool. I know is not the final and not working yet.. Cant wait until you push your changes.

One little note. On h7 the SPI_TXDR/SPI_RXDR are 32 bit register. In SPI_CFG1 can be set the length (4-32bit) of the frame. For some application it would be cool, we could set and use up to the 32bit frame (currently the functions handle 8/16 bit frames)
Not sure whether libopencm3 is universal enough to do something like this.

slorquet, 2023-01-19:
Thanks for the review, that is very useful. Also motivating to know I'm not just shouting in the dark.

I will make the functions that currently get uint16 to get uint32, and add new functions for uint16.
Also this spiv3 has stuff to pack bytes in fifo entries, I'm not sure about that. I hope defaults dont trigger a strange fifo packing behaviour.

For the moment a simple test with a SPI flash (software driven CS) stalls waiting for TXP to clear at that line:
```
while (!(SPI_SR(spi) & SPI_SR_TXP));
```

I may be missing something related to emptying/clearing fifo during init.

bm16ton, 2023-01-20:
Can confirm https://github.com/libopencm3/libopencm3/issues/1392 spi code works and finally gets my ili9341 working. Not much different from my earlier attempts except the flush function. Im software controlling cs without any issues as well. Converting to simple functions should be easy (ill get on it). ADC is next for me; bringing in adc patch set from https://github.com/allocortech/libopencm3/tree/stm32h7_adc_defs to try next.
sum clock set/gets added, external ram (needs generic functions), simple cp of i2c v2 (seems working);, spi (needs basic functions), qspi flash (needs functions, currently untested), USBHS (works), UART (works).  And all inside a single example, uhg i know, ill pretty things up once further along. I did a basic adc setup wo knowing the correct order or requirements. No errors and sum slim variances from floating but doesnt seem to be associated with PC2. Simple linux kernel driver included for easy testing. Can switch between human readable and binary output via sysfs switch. https://github.com/bm16ton/libopencm3-portentaH7

bm16ton, 2023-01-20:
Ok spi_send8 updated and spi_flush added, should any settings be added to spi_init as common use? Also adc was a stupid mistake on my part and example is working now with kernel driver. I should probly give dma a driveby. Maybe look at booting the m4 core as well.

slorquet, 2023-01-20:
flush should not be needed independently. It should be integrated in the transmit/exchange code.
Why are so many bits required to check for tx completion?

there is progress on my side: I could exchange two bytes. and now I am stuck waiting RXP to set again after trying to exchange the third byte.

My code is much simpler than yours at this point.

slorquet, 2023-01-20:
Got it working without flush. The KEY was the need to set CSTART before EACH CALL to xmit8. and define transaction len to zero AND fifo threshold to 1-data

Then you must not suspend.
This init and xfer are ok:

```
    /* Setup SPI parameters. */
    spi_reset(SPI5);
    spi_disable_crc(SPI5);
    spi_enable_software_slave_management(SPI5);
    spi_set_full_duplex_mode(SPI5);
    spi_send_msb_first(SPI5);
    spi_set_baudrate_prescaler(SPI5,2);
    spi_set_data_size(SPI5,8);
    spi_set_master_mode(SPI5);

    //spi_init_master(SPI5, 3, 0, 0, false, false);
    spi_fifo_packet_size_1data(SPI5);
    spi_set_transfer_size(SPI5,0);

    /* Finally enable the SPI. */
    spi_enable(SPI5);
```
xfer:
```
uint8_t spi_xfer8(uint32_t spi, uint8_t data)
{
    uint8_t ret;

    /* make sure data will be transmitted at next fifo write */
    SPI_CR1(spi) |= SPI_CR1_CSTART;

	/* Wait for room to write in fifo. */
	while (!(SPI_SR(spi) & SPI_SR_TXP));

	SPI_TXDR8(spi) = data;

	/* Wait for rx data in fifo. */
	while (!(SPI_SR(spi) & SPI_SR_RXP));

	/* Read the data. */
	ret = SPI_RXDR8(spi);

    return ret;
}
```

This version of the file (dont forget the header) contains working spi for h7, but DO NOT USE init_master yet. Only individual f
functions.
https://github.com/libopencm3/libopencm3/compare/master...slorquet:libopencm3:stm32h7-spi

bm16ton, 2023-01-20:
Sorry just noticed your posts, yeah that tsize is a real problem, 0 seems to be the only logical way, dont feel like turning spi on/off after every byte. Tho I have seen that done in libopencm3 books and examples as a way to control CS pin. Without tsize you loose one of the registers for knowing when transfer is complete, but txp and rxp seem to work just fine. The flush confuses me as well. I was having a byte left over in my receive fifo when doing a spi_xfer to an ili9341 that has no touchscreen so nothing to read. The xfer function seems to sending 8 then reading 8 but I was doing an additional read after just to clear that from nowhere byte. Luckily someone pointed out that PR cuz that seemed to fix it for me, for now.

Can boot m4 core now, added the following functions for that;
setcm4bootadd0
setcm4bootadd1
bootboth    (needs new name)
forcem4boot
getcm4bootadd0

bm16ton, 2023-01-20:
HA yeah Im using full duplex, Its just the xfer function on the previous stm32's would send8 read8 and all was good, But before I got it working (So I was def screwing something up) the xfer function was supposed to be sending 8 reading 8 but I still needed to do another read to empty the fifo. Probly something very foolishly silly on my part. Yeah I agree with keeping tsize 0, the errata seems to crazy otherwise. On my other older stm32's I slowdown the spi by increasing the prescaler on the fly to say take a touchscreen reading then decrease prescale on the fly before writing to screen. But I see in comments above people having issues changing settings after init. Ill hafta test that at some point could be problematic for my touchscreen en-devours.. I doubt anything I could write would be committed so no harm in setting the spi init function for h7 to tsize=0, with cs allowed to be controlled via software, and probly a few others im not thinking of, but if changes cannot be made after the fact, I may need to be more picky on that list. I guess next goal is either dma or adding m4 variant so I dont need to use arduino sketches for testing the m4. Things kinda seem like they are moving along.

slorquet, 2023-01-20:
To change settings, the spi has to be disabled.
If spi is enabled, the config registers are read only.

I got abused by spi_init_master for a long time, because this function enables spi at the end, so any config change later is not applied.

I feel this has to be changed. it should not enable spi. Or else, every function that changes settings must disable spi temporarily. this does not feel right.

slorquet, 2023-01-20:
According to RM0433 chapter 50.11.3

SPI configuration register 1 (SPI_CFG1)
Address offset: 0x08
Reset value: 0x0007 0007
Content of this register is write protected when SPI is enabled

So, yes

This is confirmed by the NuttX driver, which temporarily disables the spi peripheral to change the current data rate.

bm16ton, 2023-01-20:
Wow, been up for a couple days now, but seems like the opposite of a step forward. I see a lot of tft inits drop speed just for the init to avoid race conditions. Well ok guess I better buckup and make do. Still sucks tho. I love my f407 a little more everyday..

roman65536, 2023-01-22:
@bm16ton I tried out your libopencm3 with your examples, but looks like there are some issues with usb.
It doesn't work on my hardware (DevEBox stm32h7xx). Looks like you are using the ULPI Interface right ??

Do you know by any chance how to change it to integrated OTG HS Phy with just to wires (DP/DM) ? 

I get stucked in : 
..
halted: PC: 0x08001df2
halted: PC: 0x08001dee
halted: PC: 0x08001df0
halted: PC: 0x08001df2
halted: PC: 0x08001dee
halted: PC: 0x08001df0
halted: PC: 0x08001df2
halted: PC: 0x08001dee
^Chalted: PC: 0x08001df0
halted: PC: 0x08001df2

Program received signal SIGINT, Interrupt.
0x08001df2 in stm32f207_usbd_init () at ../../usb/usb_f207.c:91
**91	    while (OTG_HS_GRSTCTL & OTG_GRSTCTL_CSRST);**
(gdb) bt
#0  0x08001df2 in stm32f207_usbd_init () at ../../usb/usb_f207.c:91
#1  0x08001458 in usbd_init (driver=0x8004660 <stm32f207_usb_driver>, dev=dev@entry=0x80042c3 <dev>, conf=conf@entry=0x8004283 <config>, 
    strings=strings@entry=0x24000000 <usb_strings>, num_strings=num_strings@entry=0x5, 
    control_buffer=control_buffer@entry=0x240000b8 <usbd_control_buffer> "", control_buffer_size=control_buffer_size@entry=0x100) at ../../usb/usb.c:50
#2  0x080007da in main () at cdcacm.c:1061
(gdb) p OTG_HS_GRSTCTL
$3 = 0x80000001
(gdb) 

rgds
Roman

bm16ton, 2023-01-22:
Hello Roman! So your chip has the ulpi integrated? For reasons someone smarter then me would need to explain the integrated ulpi is different then external, requires registers and a clock with "phyc" in the name (if memory serves anyway). id look at Stoyan Shopov github stoyan-shopov usbhs for stm32f723. Its just sum minor changes to the init to get it going i believe. My code works on my f4s, f7s, and h7 with external ulpi, the only board i have with internal is f723 and it wont init. To switch back to using the hs port in fullspeed mode dp/dm you can copy the original init from stock libopencm3 file usb_f207.c into mine. The og driver is for that exact use case. Unfortunately im away from home/pc today (typing on broken screened phone hooefully not many typos) but when i return tonight id love to edit that init for integrated ulpi and see if we can get yeah going? If i find a min i maybe able to whio something up using my cell and post it here, fingers crossed. Wish i could see what i just ttyped i hope its legible.

roman65536, 2023-01-25:
Hallo Ben, Thank you for the hint. 
Actually after fiddling around (needed to change few things) even the usb_f107.c works. (macros need to be change to OTP_FS.. ) 
Slowly I'm making progress with the internal FS usb. There is still something wrong with the initialization of the chip (it only work when the chip had a working usb driver from stmcube and flashed with libopencm3 after it), but at least it started to communicate with me :) 
The initialization of H7 in stmcube is little bit more complex then in the libopencm3, so it needs some changes as well. 
Once it starts to work with libopencm3 from ground up, i'll let you guys know.

bm16ton, 2023-01-25:
Thats great! If i understand correctly with cubemx usb code on the board if
you place it in bootloader mode and upload a libopencm3 firmware when the
bootliader jumps it works? Makes sense, kinda like when something only
works after a redet cuz of a misorder. Im surprised the 207 driver didnt
work out of the box for yeah. I do believe i had to make 2 small if h7
changes to get usb highspeed going and one or both may be required for
usbfs over usbhs pins. Or are you using pins A11/A12 (fs pins) then def go
with 103 driver. After your last message i realized my example wont boot on
anyboard that doesnt have the same i2c device attached so i did a quick
dirty port of i2c-star (usb to i2c with linux driver support in mainline
kernel) and removed all the other stuff to make it easier/cleaner. ill do a
lil cleanup and post hopefully soonish. Remind me again your board does
have  a builtin ulpi? Thats badass hope you use that! ill help anyway i can
just ask

roman65536, 2023-01-25:
Basically I am using A11/A12 (fs pins, since those are connected to usb).  My board is a cheap H743 DevEBox board. 

Just clarify : 
1. load usb msc example build with cube.
2. connect usb -> linux recognize as it suppose to.
3. disconnect usb
4. load usb msc example build with libopencm3 
5. connect usb -> linux recognize as msc device
6. can access to device via mount/od/hexdump 
7.  back to step 3, works fine after reconnect few times 

I will go over the order how the stuff initialize (if it match with the cube order), but on this site of pond is getting late...

bm16ton, 2023-01-25:
Ive seen delays added especially for f103s and even a different h7 opencm3
usb stack, def worth a check. Also one if the common patches to get usb
going on say f7 fixes a race condition when usb isnt polled fadt enuff at
load. give me a sec ill find that otger h7 bus stahl repo.

bm16ton, 2023-01-26:
crap jist realizing never posted that link for yeah Roman my bad ill try and find it. On the I2C side of things the v2 lib can be used for a seeminly fully functional i2c bus, but I think it may be different then say the f7 i2c bus. Since i wwas porting code from f1/f4 and im new to registers and i2c i just thought any issues where do to my limited capability. But and i still need to test/play with my f7 to be sure , that differences may indeed exist. Unfortunatly i ddont have any G series or the like which seem closest to h7. The various pdf's if found from stm on migrating from one stm family to another have been woefully inadequate. Im sure they have better docs ill just hafta figure out the correct search terms or find the correct resource for finding them. That said in all its hackish glory i2c-star (well mostly) for stm32h7 is here https://github.com/bm16ton/stm32h7-i2c-star     one of the issues i have is the loop that sends each i2c byte i cannot find a register based solution for pausing until its ok to send the next one. Without any checks/pauses things get messy but seemingly (usually) later down the line in say the next i2c read (returning all zeros etc) the various register methids like use i2ctransfer and they all fail immediatly. But a simple serial write of the byte just transmitted via i2c apparently is the perfect timing but just horrendously bad coding. This particular problem i bbelieve is just me being dumb so if anyone sees where i went wrong itd be much appreciated.

bm16ton, 2023-01-29:
Added the m4 core as new variant h4, didnt know what else to do it shares nothing of peripherals with regular m4's and uses all h7's except dcache/icache and sum ram, but requires diff cflags etc then h7. But anywho it seems to be working fine. Havent looked into the 2nd set of rcc registers for it since they are only used in low power mode and thats a low priority for me.

bm16ton, 2023-02-01:
M4 support; added interrupt/usb etc and now if a peripheral is enabled by m4 core its owned by it for low power mode. I believe all that left is the hsem semaphore stuff

slorquet, 2023-03-14:
Hi,
STM32H7 also need efforts on the FLASH peripheral, which is also required for my bootloader project.
It seems that it was copied from somewhere else but never actually used.

I'll mess with it and report results.

bm16ton, 2023-03-15:
agreed i grabbed the code from somewhere(i believe in this thread) but never got around to testing it, in theory the chip specific settings ie timing need to be updated and hopefully she will take off. Please let us know if u get it working

bm16ton, 2023-03-21:
Added initial i2c fast mode plus and adc gpio companion pin switch support. On the i2c fm+ I don't have a scope etc to confirm clock speed so if anyone who does tries it please report back. Also the 20mha i2c fm+ pins aren't easy to access on my board so I didnt set them up tho I have functions in place for them. Its in my usb-2-spi/i2c/gpio/irq repo, which works amazingly well. Devices requiring irqs, and/or full duplex like mcp2515 are working fine etc. I may take this project somewhere. Currently i2c_set_speed(c->i2c, i2c_speed_fmp_1m, 8); is set to 8 but (500k ?) but works? with 16 (1mhz?) as well in my current clock tree. If not using a portenta at very least will require removing offset in ld script and first line in main. To add platform data for the spi you can either edit the spi_tiny_usb.c board_info struct and the io/properties swnode's or use a devicetree entry/overlay. The i2c uses the kernels builtin i2c-tiny-usb driver which doesn't care about speed just needs editing in firmware, in cdcacm.c in function i2c_init change the i2c_set_speed line and again in usb-2-i2c/i2c_ctx.c in function  i2c_ctx_reset again change the i2c_set_speed line.
https://github.com/bm16ton/usb-2-spi-stm32h7
crap have firmware/kernel drivers for adc I keep forgetting to add as well. and gotta get dma going. Maybe Ill  foolishly try a pr?

bm16ton, 2023-05-03:
Looks like this thread has lost momentum and Im starting to look like a crazy person, So ill leave with just a few helpful tidbits. I did get dma working, when using dma, dcache needs to be dealt with, 3 options im aware of are, Disable dcache, run a maintenance routine in every dma function, or create a protected buffer with the pmu. I added the dcache clean and invalidate function from cmsis and this did the job, but disabling dcache entirely on a dma heavy firmware seemed to perform better. Googling dcache h7 seems a lot of people dont use dcache on h7 do to hardfaults, but easy enuff to fix the hard faults are from bad error correction code bits, which get set correctly running the same scb clean and invalidate function, so just run that before configuring/enabling of dcache. Havent tried the mpu approach yet, it has restrictions and shamefully little documentation i could find. And the last thing is loop optimizations, I knew each arch had its loops optimized differently but plenty of code from my f4 and even sum from f7 (thought h7 and f7 shared core arch but i guess not) simply wouldnt work on h7 because of the various loops.  Simply using the cflag -fno-aggressive-loop-optimizations fixed all my loop issues, got to love it when its eazy. So tested working is dma, various clock settings, bdma, I added the smartcard single wire but dont have hardware to test, various other uart modes, i2c fast mode plus, the switches for the gpio companion pins attached directly to adc, booting the h4 core and a new h4 variant folder to write h4 firmware, rtc, various adc settings with and without circular dma, fsmc with portentas onboard ram, usbhs, crap I know im forgetting multiple things. Id like to try a PR but even after doing the cleaning I need to do, that will just put things in the right folders etc, wont do anything about the quality of my code. I will work on it I would like very much to contribute back anything I can.

f4grx, 2023-05-09:
I have let this project go.
I truly believe @karlp is extremely busy and has good will to help us, and I respect that but I will stop expecting anything when nothing is coming. It has been months. This project is not going forward anymore.

bm16ton, 2023-05-10:
You are spot on about Karlp's good will. I'm a dumb redneck in his 40s with a writing style of aa 3rd grader with a severe head injury. Noone in real life or forums etc has ever helped me with code except Karlp. That means a lot. So sum day when he has a call to arms to take down a repressive government or his high school bully, on that day I will f*ck that jock up. I do believe that possibly accepting pr's is difficult because the high number of boards that would need to be tested to make sure it didn't break something in code for a board the pr wasn't even meant for. I'm not sure how many, if any, other people but him actually maintain the project. With personal preferences, differing coding styles, and probly a lack of volunteers (small number anyway) I could see where bringing new people in could mean he spends as much time on each new guys commits as if he had had to do it himself. Add in a full time job and personal commitments and damn. The hard knock life of opensource devs. Since they probly only hear from people when something is broken I try and make it a point to thank them directly and explain how much it means to me. Its a far cry from what they deserve, but limited in resources am I.