libopencm3/libopencm3 #1459: Can't build project using libopencm3 and GCC version prior to 6.0

embeddedATsleepinnov, 2023-01-23:
After library update from a very old version (approx. 2013), my project doesn't build anymore.

Since previous version, lot of changes have been done so I migrated my project to use current API.
But I can't build my project using libopencm3 source as is. To get it building, I had to remove `__attribute__ ((deprecated()))` used in include/libopencm3/stm32/common/i2c_common_v1.h.

The reason is because GCC support deprecated attribute in enum since version 6.0 and for some reasons I use GCC 4.6.2.

I resolved my problem by

- replacing `__attribute__ ((deprecated()))` by `LIBOPENCM3_DEPRECATED_UNION()` in i2c_common_v1.h
- adding following code to include/libopencm3/cm3/common.h
```
#ifdef __GNUC__
#	if __GNUC__ > 5 || (__GNUC__ == 5 && __GNUC_MINOR__ > 5)
#		define LIBOPENCM3_DEPRECATED_UNION(x) LIBOPENCM3_DEPRECATED(x)
#	else
#		define LIBOPENCM3_DEPRECATED_UNION(x)
#	endif
#else
#	define LIBOPENCM3_DEPRECATED_UNION(x)
#endif
```

Hope it help!

karlp, 2023-01-23:
There's no real interest in supporting that old a gcc in the current library I'm afraid.

embeddedATsleepinnov, 2023-01-24:
In this case it should be great to update Toolchain section of the README. The NOTE recommends to use _gcc-arm-embedded version 4.8 2014q3 or newer_ but it may not build on some target using gcc prior to 6.0.

karlp, 2023-01-24:
That's totally fair.  4.8 was required to get the cortex-m7 support at one point.  I'll update the readme.

closed 2023-01-24 by karlp