libopencm3/libopencm3 #1462: Building issue appears(Syntax Error)

Unlimitosu, 2023-01-30:
I tried to build this with `make` command in wsl, Syntax error is occured.

The error description is as follows:

> BUILD   lib/stm32/f0
> make --directory=lib/stm32/f0 SRCLIBDIR="/home/unlimit/arm/polymul-z2mx-m4/libopencm3/lib\" || \
>         echo "Failure building: lib/stm32/f0: code: $?" > .stamp_failure_lib_stm32_f0
> /bin/sh: 2: Syntax error: Unterminated quoted string
> make: *** [Makefile:69: lib/stm32/f0] Error 2

The description is generated by `make  V=1` command.

I think there is some problem with the Makefile, but I'm not sure.

What should I do to solve this issue?

+ I get the repository by `git submodule update --init` command after I cloned [this](https://github.com/mupq/polymul-z2mx-m4). Then I ran `make` command in the libopencm3 directory, and the error occured.

Unlimitosu, 2023-01-30:
@karlp Sure, I ran 'make -v' and the result is as following:

> GNU Make 4.3
> Built for x86_64-pc-linux-gnu
> Copyright (C) 1988-2020 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.

karlp, 2023-01-30:
vmware running what OS? vmware with a "general" linux should work just fine, I don't have any experience with WSL though sorry.

Unlimitosu, 2023-01-31:
OS is Windows 10 and there's still same errors with wsl. And it solved after I use vmware with Ubuntu 20.04(I used 22.04 initially). Thanks.

closed 2023-01-31 by Unlimitosu