libopencm3/libopencm3 #1467: rcc_clock_setup_pll() does not allow switching oscillators

jsphuebner, 2023-02-06:
Maybe this is not a bug in opencm3 but intended hardware behaviour?

Tested on STM32F103

I have written a boot loader that initializes the clock using
`rcc_clock_setup_pll(&rcc_hsi_configs[RCC_CLOCK_HSI_24MHZ]);`

It is supposed to be universal.
If the firmware uses that same clock setup all is well. But if it uses a different one or at least a different clock source it seems to not work.

`rcc_clock_setup_pll(&rcc_hse_configs[RCC_CLOCK_HSE8_72MHZ]);`

The toggle LED then blinks at around 1/5 of the intended 100ms rate. Here's the bootloader
https://github.com/jsphuebner/stm32-CANBootloader/

karlp, 2023-02-06:
I'm going to call it intended behaviour.

What you've done with your bootloader is handover to the app in a very unclean state.  You can define that as your contract with the apps, but it's obviously causing you problems :)

One method is to use a static ram variable, and use sysreset to boot cleanly, jumping immediately based on the ram flag.

jsphuebner, 2023-02-06:
Ok, thanks!
So clock_setup...() most only be called once after reset then

karlp, 2023-02-07:
I mean, it probably _should_ work, if you can find a bug in it, go for it, but we've already found cases where it won't work, because going up/down speeds need to do things in different orders. See https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/l1/rcc.c#L484-L498 for wher eit's been handled, just for one case, but should probably be handled for more cases?

jsphuebner, 2023-02-09:
I've now done a more subtle clock setup:
```
   rcc_set_pll_multiplication_factor(RCC_CFGR_PLLMUL_PLL_CLK_MUL6);
   rcc_set_pll_source(RCC_CFGR_PLLSRC_HSI_CLK_DIV2);
   rcc_osc_on(RCC_PLL);
   rcc_wait_for_osc_ready(RCC_PLL);
   rcc_set_sysclk_source(RCC_CFGR_SW_SYSCLKSEL_PLLCLK);
```
I don't worry about the other prescalers and setting apb2Clock etc. as that is not being used.
But most of all I'm restoring the clock source to unmultiplied HSI and turn off the PLL before leaving the boot loader:
```
   rcc_set_sysclk_source(RCC_CFGR_SW_SYSCLKSEL_HSICLK);
   rcc_osc_off(RCC_PLL);
```
Now the application can successfully switch to a different clock source