libopencm3/libopencm3 #1469: Request for ASM friendly headers

ppbrown, 2023-02-13:
I wanted to write some ASM for a few STM32 boards I have in my posession.
I came across this project as a good sort of code but...
I had to hand-rewrite everything to be asm compatible.
Which really shouldn't be necessary for a lot of the code.

It would be really nice if:

1. the base entrypoint and vector table type stuff, were generated in .s format, instead of .c format.

2. where possible, the defines are made to be ASM friendly.
with perhaps the non-ASM friendly bits being put in #ifdef blocks,to make it clear which ones are usable in both modes, vs which ones are C-only.

A reminder that with the GNU toolchain, you can actually use constant numerical defines(and even some more complex ones) directly in ASM, as long as the programmer names their files as "file.S" instead of "file.s"

karlp, 2023-02-13:
1: Absolutely not.  There's no reason at all to do startups in assembly, it was a banner feature of cortex-m that it was no longer required, and there's no way that we'll revert to stone age tooling like that.

2:  What do you need concretely different?  To a certain point, things can be accomodated, but as you might be able to infer from (1) we're not intending to ever be a "use this if you want to work in assembly"    Are your needs not met by just using the raw CMSIS headers themselves?

ppbrown, 2023-02-13:
Hmm.  Interesting!
For 1, I had incorrectly assumed that linking in libopencm3_stm32.a would pull in a lot of junk. However, after reexamination, that seems to be not quite the case. 
Technically, there IS a certain amount of junk: initialization of C++ stuffs that will never be used. I'm not happy about that. But at least its relatively small.

On the other hand, For 2, some of the defines are rather geared towards use with the C functions in your library, rather than just providing hardware-representative offsets.
Specific example

include/libopencm3/stm32/f4/rcc.h
I want to be able to use defines related to enabling GPIOA clock in the appropriate RCC register.
However, RCC_GPIOA is ... an *enum* ?!?

Not ASM friendly, last I checked :-D

ppbrown, 2023-02-14:
I notice that this seems to ORIGINALLY have been a value of the project.
For example, 

include/cm3/common.h`

has things like

#if defined (__ASSEMBLER__)

ppbrown, 2023-02-14:
I COULD submit a PR, for the majority of rcc.h files under stm32, that would add a defines for RCC GPIO enabling, for ease of use in ASM.

But it would not be present for the sub-arches that are ill-behaved

Example:

f4/rcc.h would have

#define RCC_GPIO_ENR  RCC_AHB1ENR

l4/rcc.h would have

#define RCC_GPIO_ENR  RCC_AHB2ENR

But f1/rcc.h and f3/rcc.h would not, because unlike the others, the enable bit for GPIOA, is not bit 0
Probably I would add a comment to that effect for those oddbalsl.

How does that sound?

karlp, 2023-02-14:
For the CM3 common files, it's worth bearing in mind that when this project started, the original ARM cmsis headers were _not_ usable under any sort of free software license.  they're  now apache2, so, in very real terms, a LARGE chunk of the header definitions in this repository are of questionable value, other than being simply a different "shape" to the CMSIS headers.  

As for your latter proposale, to add RCC_GPIO_ENR redirections, I'm failing to see how that relates to some sort of assembler requirement?  You just seem to want to add a HAL wrapper layer?  I'm personally very uninterested in adding another partial attempt at that.  Where would you stop?  The define soup is already error prone and hard to refactor, throwing more at it doesn't sound like any sort of good idea.

ppbrown, 2023-02-14:
(FYI: You earlier asked, "Are your needs not met by just using the raw CMSIS headers themselves?"
They are not. The headers intermingle use of typedef and struct in them, so are unusable for ASM.)

I see my proposal not so much adding a HAL layer, as translating the existing one.
rcc_periph_clock_enable(), with its matching enum, IS a hardware extraction layer.
Sadly, the enum makes the definitions unusable in ASM, so i'm proposing adding a small translation.

It turns out that, for the basic "get a light to blink/read a button press" type demos, the existing libopencm3 headers do serve the purpose... except for this ONE THING, of needing to have a reference to the correct RCC register for enabling GPIO.

ppbrown, 2023-02-14:
ah nuts.

I take it back.
when I actually test my proposal, I find that because of all the included other files, libopencm3 suffers from similar problems.
I'm a little sad there is no pre-existing easily modifyable solution for this.

oh well.
Thanks for the replies.

closed 2023-02-14 by ppbrown

karlp, 2023-02-15:
given that the headers are mostly static once created, a simple fork and delete the undesired stuff is _probably_ pretty easy to maintain.  You may also like to look at some of the SVD based header generators.  Though, many of them are also targetting high level languages, so are highly likely to also include niceities for high level languages.  Good luck!

ppbrown, 2023-02-15:
if i did fork, though, it wouldn’t be just a fork and delete. there are a
lot of inconsistencies even just within the stm32 subdirectories.
use of the BIT0,1,2 macros vs reinventing them. even for literally the same
purposes.
And that’s just one example.