libopencm3/libopencm3 #1473: Please take care of open issues

slorquet, 2023-03-08:
Hello,
Please advise on how to make progress with stm32h7

Last time I checked you refused to merge our code but did not provide any way to go forward and make it mergeable. It has been one full month.

Thank you for your help

karlp, 2023-03-08:
I didn't refuse to merge anything.  You are simply _wildly_ overestimating both a) my free time that I can give to this project and b) your ability to control that time.  I actually went and bought a h7 board and have it plugged in on my desk specifically to try and help get the h7 support moved along.   I'm sorry that this project does not move at the velocity you desire, but that's the reality on the ground.

closed 2023-03-08 by karlp

slorquet, 2023-03-08:
Hello,

I am highly respectful that you are extremely busy and I am not asking you to do things you cannot do.

I am just wondering how I can help and I am sorry I cannot. Also I am a moron and I should have just asked you how could we progress forward instead of just telling you that we are late.

Also I know that STM32H7 boards are hard to find.

Thats good news if you could get a board, please dont hesitate to ping so I can fix what need to be fixed.

You have a very clear idea of what you want in this project, much clearer than any of us, so I cannot make much progress until you process what we sent.

Thank you and all the best for whatever eats your free time.

karlp, 2023-03-08:
It's ok, no hard feelings, and you're not a moron, don't got there.  I'm the one continually letting people down by not getting to all of their code.  I try and respond promptly early with anything major so that when I can find time, I can "just work through" the rest of it.  but that does sometimes give the illusion of more responsiveness that is really there :)