libopencm3/libopencm3 #1475: STM32F302 USB Issue

arendwierks, 2023-03-24:
Hello,

I am having a issue with the USB library using a STM32F302 device.

When I compile and upload the [usb-cdcacm](https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f3/stm32f3-discovery/usb_cdcacm) example to my board Windows doesn't recognize the device.
I also have an STM32F303 board, and the example works fine on that board.
To check if my board was working fine I made a project with STM32CubeIDE, and when I upload that to my STM32F302 board Windows recognizes the board as an serial port device.

So I think the 302 USB initialisation is different in a little way from the 303. And libopencm3 doesn't seem to handle this difference correctly.

Does anybody else run into this issue of does anyone have solution?
Thanks in advance.

manuelbl, 2023-03-24:
To the best of my knowledge, the STM32F302 and STM32F303 differ in the *access scheme* for the USB buffer descriptor table:

- The STM32F303 has 1024 bytes of RAM for USB buffers in a range of 1024 bytes (0x40006000 – 0x400063FF).
- The STM32F302 has 512 bytes of RAM in the same range of 1024 bytes (0x40006000 – 0x400063FF), but the upper half of each word cannot be used. This access scheme is also used by the STM32F1 family and possibly further MCUs.

The difference can be seen in `st_usbfs_copy_to_pm` (https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/st_usbfs_v1.c#L57 vs. https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/st_usbfs_v2.c#L44), in `st_usbfs_copy_from_pm` in the same files.

*libopencm3* is configured to use `st_usbfs_v1.c` for the entire STM32F3 family. But the STM32F302 would actually require a mix between `st_usbfs_v1.c` and `st_usbfs_v2.c`.

karlp, 2023-03-27:
I think one idea had been to just compile in st_usbfs_v1.o and st_usbfs_v2.o in the f3 makefile, but I didn't have sufficient boards to test that.  if that lets you make code for both boards, excellent! let us know.  Alternatively, just replace v1 with v2 when you need the other target.  Again, I don't have both variants to test that one.  Just edit https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/f3/Makefile#L60

karlp, 2023-03-27:
see also https://github.com/libopencm3/libopencm3/issues/841

manuelbl, 2023-03-27:
The STM32F302 uses the USB buffer memory layout like as in st_usbfs_v2.c. But it doesn't have all the registers, e.g. USB_BCDR is used in st_usbfs_v2.c but not available in the STM32F302.

So the st_usbfs_v2.c driver might. Or it might require a slightly modified variant.

PocketPi, 2023-08-21:
I think i have the same problem with a 411 vs 412 could anyone confirm that?