libopencm3/libopencm3 #1477: nvic.h - again :-(

Ekkehard, 2023-04-03:
I saw quite a few posts here about the missing nvic.h file.  Unfortunately, none of the recommendations there helped :-(

I just downloaded a brand new copy of libopencm3.  My toolchain is arm-gnu-toolchain-12.2mpacbti-rel1 running under Windows.

I am using GNU make 3.81.

I get the identical behavior on Windows 10 and Windows 11.  It works fine on my Mac with the same toolchain and make version.

Here are the three commands that I issued under Windows that lead to the problem I am facing

prompt>**git clone https://github.com/libopencm3/libopencm3.git**
Cloning into 'libopencm3'...
remote: Enumerating objects: 30568, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 30568 (delta 5), reused 13 (delta 5), pack-reused 30550
ts: 100% (30568/30568), 6.51 MiB | 1.43 MiB/s
Receiving objects: 100% (30568/30568), 7.01 MiB | 1.45 MiB/s, done.
Resolving deltas: 100% (19937/19937), done.

prompt>**cd libopencm3**

prompt>**make TARGETS=stm32/f1**
  GENHDR  stm32/f1
make: *** [include/libopencm3/stm32/f1/nvic.h] Error -1073741502

prompt>

And indeed, nvic.h hasn't been created in the f1 directory.

Of course, I could just copy the nvic.h file from my Mac over to the Windows boxes, but I am afraid that this might cause even more problems down the road - after all, something must have gone wrong, and I'd like to fix it before I move on.

Any help would be highly appreciated!!

karlp, 2023-04-03:
run ```make V=1``` and check that you have a "python3" binary installed and usable when you are at that prompt.

Ekkehard, 2023-04-03:
I get (on the more cooperative Windows 10 box - Windows 11 I first had to create a link from python3.exe to py.exe)
prompt> **where python3**
C:\Users\ekkehard\AppData\Local\Microsoft\WindowsApps\python3.exe

and

prompt> **make V=1**
make -C libopencm3 TARGETS=stm32/f1
make[1]: Entering directory `D:/Users/ekkehard/EmbeddedSystems/STM/STM32F/libopencm3'
  GENHDR  stm32/f1
./scripts/irq2nvic_h ./include/libopencm3/stm32/f1/irq.json;
make[1]: *** [include/libopencm3/stm32/f1/nvic.h] Error -1073741502
make[1]: Leaving directory `D:/Users/ekkehard/EmbeddedSystems/STM/STM32F/libopencm3'
make: *** [libopencm3/lib/libopencm3_stm32f1.a] Error 2

Any more ideas?

Ekkehard, 2023-04-03:
OK Karl - I got it resolved - user error (as usual).  I overlooked the tiny little fact that I was supposed to run the make command in the msys shell not in a PowerShell or DOS shell as libopencm3 obviously also uses bash shell scripts.

Never ceases to amaze me: The minute you do it right - it works!!

Shame on me for not using the msys shell!!

closed 2023-04-04 by karlp