libopencm3/libopencm3 #1487: Allowing bootloaders in genlink.py or device.data

dxhisboy, 2023-07-06:
I'm currently using the makefile modules in the `mk/` directory. But I cannot find a way to supply additional offset of ROM when using a DFU bootloader.

I suggest adding a way to override `-DROM_OFF` provided by `genlink.py` when generating ldscript or override that variable in `genlink.py` internally.

Possible solutions are:

1. Adding `.dfu` suffixes in `DEVICE`, and `genlink.py` handles the flag.
2. Adding `,ROM_OFF=xxx,RAM_OFF=xxx` as a suffixes in `DEVICE`, `genlink.py` build a dict and override corresponding variables internally after reading out device data.
3. Allowing to pass or override those flags when calling `cpp` in ldscript generation.

If any way above is allowed, I could implement corresponding functionality and initiate a PR.

karlp, 2023-07-06:
Personally, I think the right option here is to provide your own linker header, that includes the common one, rather than making more complicated magic. An example for a libopencm3 project that uses a bootloader is as simple as
```
MEMORY
{
	rom (rx) : ORIGIN = 0x08002000, LENGTH = 56K
	ram (rwx) : ORIGIN = 0x20000000, LENGTH = 32K
}

INCLUDE ../common/cortex-m-generic.ld
```
(cortex-m-generic.ld is whatever path you need to get to https://github.com/libopencm3/libopencm3/blob/master/lib/cortex-m-generic.ld)
If that's sufficient for you, you can close this.  If you really want to do more with this, leave this open and see if anyone else has an opinion on it.

dxhisboy, 2023-07-07:
> rather than making more complicated magic.

I understand, but I still want the compile flags from `genlink.py` since I'm not familiar to ARM MCUs. Maybe I should call `genlink.py` in my Makefile to acquire this in another way.

Thank you.

closed 2023-07-07 by dxhisboy

karlp, 2023-07-07:
If you include genlink-config, but just skip genlink-rules, you get all the arch flags you want, you just don't get the LDSCRIPT defined. and you can then just point at your own as I suggest.