libopencm3/libopencm3 #1488: Question about STM32WLxx support

virtualabs, 2023-07-06:
I am currently working on adding support for the STM32WLxx family MCUs into libopencm3, especially the STM32WL55 and STM32WLE5 variants. Since these MCUs have sub-GHz integrated transceivers (LoRa/FSK/MSK/BSK/Sigfox capable), I was wondering if it makes sense to include a sub-GHz driver into the libopencm3 project, regarding a future pull request. 

This is still a work in progress, but I got almost everything working fine (including modems) with this family of MCUs :)

bm16ton, 2023-07-23:
Anychance you have the source posted anywhere?

virtualabs, 2023-07-24:
I've forked the `libopencm3` repository and added this new `wl` variant, the code is available here: https://github.com/virtualabs/libopencm3/tree/new-arch-stm32-wl/lib/stm32. My sub-GHz driver is still in development and has not been integrated to libopencm3 yet, but can be found in this other repository: https://github.com/virtualabs/3615-lora-firmware.

bm16ton, 2023-07-24:
Awesome! A very quick search made it seem like I kay have troubles finding one of these boards..at least for a reasonable price, but it was a quick search. This is great ill keep an eye on this and add it to my usbhs/h4/h7 repo. I've wanted to try one of these dual radio W variants for a while but availability has been an issue. But this motivates me. Thank you sir

virtualabs, 2023-07-24:
SeeedStudio is selling a board based on a STM32WLE5JC which may be compatible for about USD 20 (https://www.seeedstudio.com/LoRa-E5-mini-STM32WLE5JC-p-4869.html), but it does not expose all the pins and therefore cannot be used to debug the internal SUBGHZ_SPI bus. I've tried running a sample firmware on one of these and it seems fine, but note sure if the radio driver works as expected.

bm16ton, 2023-07-24:
That looks very promising! Thank you very much, guess I could coble together some other way of debugging. I assume the big issue isn't being able to sniff the spi via headers? But the price is right

bm16ton, 2023-07-24:
Oh ok this is a single radio version. Am I mis remembering or do they have a dual radio W series with 2.4ghz and sub ghz radios? Its nice how they repeatedly say in write up it breaks out all gpios/peripherals, and I very much believe you that they dont. I hafta google now for the dual radio models.