libopencm3/libopencm3 #1496: Support for RP2040 ARM Cortex-M0+ MCU

hattesen, 2023-09-03:
Hi,

Is there any plans to add support for the (popular and low cost) [RP2040](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) ARM Cortex-M0+ MCU?

Does a "Porting Guide" exist, that explains the process/steps involved in porting libopencm3 to a new ARM Cortex architecture?

karlp, 2023-11-05:
because when libopencm3 started, the cmsis headers weren't permissively licensed, like they are now.

karlp, 2023-11-05:
As for the issue itself, there's no porting guide, and people who want RP2040 are welcome to try and submit support for it.  I am not personally working on it.  We also don't really keep "wishlist" tickets open, they just get in the way of real issues, so I'm closing this for now.

closed 2023-11-05 by karlp