libopencm3/libopencm3 #1501: DMA for STM32H7

srobertjames, 2023-09-27:
Does libopencm3 provide DMA for STM32H7? The includes seems to miss it:

```
$ cat libopencm3/include/libopencm3/stm32/dma.h 

#if defined(STM32F0)
#       include <libopencm3/stm32/f0/dma.h>
#elif defined(STM32F1)
#       include <libopencm3/stm32/f1/dma.h>
#elif defined(STM32F2)
#       include <libopencm3/stm32/f2/dma.h>
#elif defined(STM32F3)
#       include <libopencm3/stm32/f3/dma.h>
#elif defined(STM32F4)
#       include <libopencm3/stm32/f4/dma.h>
#elif defined(STM32F7)
#       include <libopencm3/stm32/f7/dma.h>
#elif defined(STM32L0)
#       include <libopencm3/stm32/l0/dma.h>
#elif defined(STM32L1)
#       include <libopencm3/stm32/l1/dma.h>
#elif defined(STM32L4)
#       include <libopencm3/stm32/l4/dma.h>
#elif defined(STM32G0)
#       include <libopencm3/stm32/g0/dma.h>
#elif defined(STM32G4)
#       include <libopencm3/stm32/g4/dma.h>
#else
#       error "stm32 family not defined."
#endif
```

If yes: How do I use it?
If not: Why not? Is there a workaround (or is it currently: "If you need DMA on H7, don't use libopencm3?") What's involved in porting it?

If we need to use ST32 LL/HAL for this, is there a way to do so and still use libopencm3 for other functionality?

rocpe, 2024-04-27:
i2c seems to be also missing.

karlp, 2024-04-29:
new parts get support added when people test and add that support.  Sometimes this as simple as adding common objects to makefiles and expanding the list you see above, sometimes it's more than that.  However, we're not keeping issues open for each and every missing thing.  It implies that the reported missing is the only missing, but we're likely never feature complete.

closed 2024-04-29 by karlp