libopencm3/libopencm3 #1504: STM32L496 USB CDC-ACM

ALSCode, 2023-10-22:
Hello!
Failed to create and launch simple usb cdc-acm example for stm32l4 mcu. Host (linux and windows) don't see usb devices.  As a template I used cdc-acm example for stm32f407 mcu from PlatformIO framework. Please, see code attached. Could someone take a look and point, what I do wrong. Thanks!
[cdcacm.c.tar.gz](https://github.com/libopencm3/libopencm3/files/13064345/cdcacm.c.tar.gz)

ChuckM, 2023-10-22:
The L496 uses a different version of the USB peripheral than the 407 does. See the examples repo for the other 4L variants, do not use examples from the F4 variants.

ALSCode, 2023-10-23:
> The L496 uses a different version of the USB peripheral than the 407 does. See the examples repo for the other 4L variants, do not use examples from the F4 variants.

Hello, ChuckM!
Thank you for fast reply! Yes, I use recommended for L4  usb driver  st_usbfs_v2.c. Unfortunately, without any success. And about second part of your answer - as I see, there are no libopencm3 USB examples for stm32L series at all! That's why, I ask for help and guidance