libopencm3/libopencm3 #1509: API differs significantly between target microporcessors

riban-bw, 2023-11-06:
The library is a lightweight wrapper for low-level (bare metal) registry access with helper functions to ease development. It therefore needs to be flexible in how it exposes core functionality but it also has an API that users may expect to be as consistent as possible. This is not the case. Performing the exact same operations on very similar microprocessors is implemented substantailly differently. For example, to configure a pin as an output is done in these different ways:

```
// STM32F0
void gpio_mode_setup (uint32_t gpioport, uint8_t mode, uint8_t pull_up_down, uint16_t gpios)
void gpio_set_output_options (uint32_t gpioport, uint8_t otype, uint8_t speed, uint16_t gpios)

// STM32F1
void gpio_set_mode ( uint32_t gpioport, uint8_t mode, uint8_t cnf, uint16_t gpios)
```
These both do the same thing but are presented differently for each microprocessor for no apparent reason.

Mnay constants are also named significantly differently for similar or identical uses and values.

If a user is designing for just one microcontroller then these differences are not obvious but as soon as they use multiple processors of try to port code between processors they hit this challenge. Common code becomes very messy with lots of preprocessor directives just to define function calls and constants that ultimately do the same thing.

It appears that the codebase has evolved, possibly with multiple authors or an author developing and improving ideas. I suggest that the framework would be more accessible if it were more consistent. I have been developing for just two targets (STM32F0 & STM32F1) which are very similar in core architecture but I have struggled to comprehend the APIs that differ between the two. My code ends up being twice the size (due to separate implementation of the same code for each processor) or confusingly complex (due to preprocessor directives) which slows development and adds to the techincal overhead / debt.

I appreciate how challenging it would be to resolve this issue, especially whilst maintining backwards compatibility but it does present significant friction to users. If there were a definitive template for functions and constants that could be referenced then it may be easier for further development (and retrospective fixes) to be implemented, e.g. speficy which API is the reference one and use its naming conventions / function templates / constants, etc. as the basis for others.

karlp, 2023-11-06:
Correct, and an ongoing process when work is done.  However, this project was never attempting to be a portable HAL layer, but rather (at the time) a freely licensed manner of working with the hardware.  Many other projects work on HAL compatible layers, and providing both layers in this repository is simply out of scope.  What _is_ fixed when it's found, is where the same peripheral is used in different parts within the same family, but has different names.  All of those incompatibilities we try and fix.  Your specific example about f1 gpio is the reason why most people don't use the f1.  It is the _only_ stm32 that has that gpio style, because that's the api of that gpio peripheral.  There's no way around that without a second layer on top.  However, again, we're not keeping non-specific wishlist tickets open for anything here.

closed 2023-11-06 by karlp

riban-bw, 2023-11-07:
Understood, thanks! I completely understand and agree. I will report specific issues in the future that detail single individual differences and will endeavour to offer PRs what I can 

It may be worth adding a comment to the README that describes the current maintenance status of the project. It is reassuring to get such prompt responses to issues but I get the impression that there may be limited resources (effort) being targeted at the project. It may help users understand what they may expect in the way of support and may garner offers of assistance. Cheers!