libopencm3/libopencm3 #1514: STM32G0: #include <libopencm3/cm3/nvic.h> leads to a NON existing [libopencm3/stm32/g0/nvic.h]

zinahe, 2023-11-15:
Hi,

I'm trying to build a relatively simple program on STM32G030 which includes `<libopencm3/cm3/nvic.h>` that ends up in a `dispatch/nvic.h` which then leads to a NON existing `g0/nvic.h`as shown below:

https://github.com/libopencm3/libopencm3/blob/1f3abd43763fa39d23e737602b6d0011a45c70b2/include/libopencm3/cm3/nvic.h#L147

https://github.com/libopencm3/libopencm3/blob/1f3abd43763fa39d23e737602b6d0011a45c70b2/include/libopencm3/dispatch/nvic.h#L23-L24

I believe this file used to exist in the past, but was deleted at some point.

Regards,

Zinahe A.

zinahe, 2023-11-15:
Yes. I'm an idiot. 

I just cloned the repo and proceeded to compile my code against libopencm3. Didn't even bother reading the README. 

Running `make targets='stm32/g0'` solved the issue.

closed 2023-11-15 by zinahe

karlp, 2023-11-15:
glad you got things sorted out :)