libopencm3/libopencm3 #1515: Not able to compile using STK_CSR_CLKSOURCE_AHB_DIV8 on STM32G0

zinahe, 2023-11-15:
Hi,

I get a compile time error when I use `STK_CSR_CLKSOURCE_AHB_DIV8` on STM32G0 even though the reference manual says I can. The following C code:

`systick_set_clocksource(STK_CSR_CLKSOURCE_AHB_DIV8);`

results in a compile time error:

`main.c:34:26: error: 'STK_CSR_CLKSOURCE_AHB_DIV8' undeclared (first use in this function); did you mean 'STK_CSR_CLKSOURCE_AHB'?`

And when I call the function as follows which means exactly the same, it compiles with no issues and the MCU also behaves as expected

`systick_set_clocksource(0 << STK_CSR_CLKSOURCE_LSB);`

https://github.com/libopencm3/libopencm3/blob/1f3abd43763fa39d23e737602b6d0011a45c70b2/include/libopencm3/cm3/systick.h#L119-L120

karlp, 2023-11-16:
right, but look at the #ifdefs on arch armv6m right there.    cortex-m0's _don't_ have a "ahb div 8" bit.  they hav ea "external clock" bit.  On stm32's, this has been configured, _BY ST_ to "do the same thing" as on their m3/m4 parts.    The defines you're trying to use are from the _cortex_ part, not the ST parts.

This is handled in the documentation for you, see : http://libopencm3.org/docs/latest/stm32g0/html/group__CM3__systick__defines.html#ga13726e0964b31c85269beddbdc1d35e7 where it provides a link to the options available on the part you're using:
http://libopencm3.org/docs/latest/stm32g0/html/group__systick__clksource.html

closed 2023-11-16 by karlp

zinahe, 2023-11-17:
@karlp, thanks for the quick response.

So if I understood you correctly, from the point of view of ARM's Cortex-M0/+ specification there is no AHB_DIV8 but either AHB or EXT when selecting a clock source for SYSTICK. But ST went ahead and implemented an AHB_DIV8 feature anyway for their stm32g0 but now we have to refer to it as EXT since we are using ARM's include file i.e. `systick.h`. Right ?

karlp, 2023-11-17:
ST connected AHB_DIV8 to the EXT input, yes.  so yes, on cortex-m0, to use cortex-m0 standard functions, you need to ask for "external clock" to systick...