libopencm3/libopencm3 #1538: make[1]: arm-none-eabi-gcc: No such file or directory

justinjools22, 2024-04-06:
I am unable to compile with Make. I'm using Msys2 64 bit Windows 11. I have tried this several times, and searched for a solution.

make[1]: arm-none-eabi-gcc: No such file or directory
make[1]: *** [../../Makefile.include:41: adc.o] Error 127
  BUILD   lib/stm32/f1
  CC      adc.c

Failure building: lib/efm32/ezr32wg: code: 2
Failure building: lib/efm32/g: code: 2
Failure building: lib/efm32/gg: code: 2
Failure building: lib/efm32/hg: code: 2
Failure building: lib/efm32/lg: code: 2
Failure building: lib/efm32/tg: code: 2
Failure building: lib/efm32/wg: code: 2
Failure building: lib/gd32/f1x0: code: 2
Failure building: lib/lm3s: code: 2
Failure building: lib/lm4f: code: 2
Failure building: lib/lpc13xx: code: 2
Failure building: lib/lpc17xx: code: 2
Failure building: lib/lpc43xx/m0: code: 2
Failure building: lib/lpc43xx/m4: code: 2
Failure building: lib/msp432/e4: code: 2
Failure building: lib/nrf/51: code: 2
Failure building: lib/nrf/52: code: 2
Failure building: lib/pac55xx: code: 2
Failure building: lib/sam/3a: code: 2
Failure building: lib/sam/3n: code: 2
Failure building: lib/sam/3s: code: 2
Failure building: lib/sam/3u: code: 2
Failure building: lib/sam/3x: code: 2
Failure building: lib/sam/4l: code: 2
Failure building: lib/sam/d: code: 2
Failure building: lib/stm32/f0: code: 2
Failure building: lib/stm32/f1: code: 2
Failure building: lib/stm32/f2: code: 2
Failure building: lib/stm32/f3: code: 2
Failure building: lib/stm32/f4: code: 2
Failure building: lib/stm32/f7: code: 2
Failure building: lib/stm32/g0: code: 2
Failure building: lib/stm32/g4: code: 2
Failure building: lib/stm32/h7: code: 2
Failure building: lib/stm32/l0: code: 2
Failure building: lib/stm32/l1: code: 2
Failure building: lib/stm32/l4: code: 2
Failure building: lib/swm050: code: 2
Failure building: lib/vf6xx: code: 2
make: *** [Makefile:80: lib] Error 1

karlp, 2024-04-06:
You need a toolchain.  Line number one of your error says
`make[1]: arm-none-eabi-gcc: No such file or directory`

It can't find the toolchain.  A toolchain is also listed in the prerequisites: https://github.com/libopencm3/libopencm3/?tab=readme-ov-file#prerequisites

justinjools22, 2024-04-06:
I installed Arm GNU Toolchain to the root of Msys2 and used this path. Works.

export PATH="$PATH:/Arm GNU Toolchain arm-none-eabi/bin/"  
cd libopencm3
make