libopencm3/libopencm3 #1541: Configurable constness of USB descriptors.

amalinowski75, 2024-04-17:
I've hit a problem with "constness" of field of structs of USB descriptors. In particular I had a problem with 

`const struct usb_endpoint_descriptor *endpoint;` in usb_interface_descriptor struct in usbstd.h file.

The "const" qualifier prevents me from reconfiguring usb device in runtime. I need it as I need to change the configuration of composite devices (change their orders and the set of devices) on demand during the firmware execution. I overcame the problem by removing the qualifier from my copy of libopencm3 (submodule) but it's not an ideal solution.

I know the const qualifiers are in the code not without a reason but maybe I'm not alone and there are other people that need to have the structures in RAM to modify their fields and maybe we could make the constness configurable.

What do you think about that @karlp ?

karlp, 2024-04-18:
Personally, I don't think it sounds like a good way to go.  There's another PR that actually goes _further_ to const the descriptors, so they can be entirely in flash, instead of being constructed into ram as they are now.    (this is a big improvement for ram usage) But, I'm not goin to be doing any major work personally anyway, so see what other support you can drum up if you think this is a meaningful case.  (You're the first person I've seen request it though, so perhaps consider if you can do this in other ways? pointing to alternate consts perhaps?)

amalinowski75, 2024-04-18:
Yeah, I've seen this PR and I'm pretty aware of the fact that vast majority of developers are not interested in changing USB descriptors in runtime. I'm happy to be the first one here :)
I was thinking about creating a macro that could be either 'const' or empty depending on some definintion in Makefile.
I know people hate macros but that sounds like a pretty easy change. Of course it may be easy at first glance but in reality it may turn out to be a train of questions like which consts to macro-ize?, should be there some groups of cosnts, etc.
Anyway, at this point I'm going to create a fork of libopencm3 and use it for my project. Maybe I will come up with some PR for that in the future.
All I wanted was to see what's your opoinion on this kind of change and now I know it's not a song of the near future :)